kevancress/MeasureIt_ARCH #1: Installation problem

Blenderbugg, 2019-03-25:
Hi,
getting the following error when I tried to activate addon into latest build of blender 2.8

Modules Installed () from '/Users/newuser/Downloads/BLENDER/2.8 addons/archectural/measureit_arch_v01.zip' into '/Users/newuser/Library/Application Support/Blender/2.80/scripts/addons'
Traceback (most recent call last):
  File "/private/var/folders/k6/1x1hxh5d6bs54nmjxb030mr00000gn/T/AppTranslocation/283ADC70-B960-4CC9-8C28-6F0E0F4DEBEC/d/blender.app/Contents/Resources/2.80/scripts/modules/addon_utils.py", line 351, in enable
    mod = __import__(module_name)
  File "/Users/newuser/Library/Application Support/Blender/2.80/scripts/addons/measureit_arch_v01/__init__.py", line 84, in <module>
    auto_load.init()
  File "/Users/newuser/Library/Application Support/Blender/2.80/scripts/addons/measureit_arch_v01/auto_load.py", line 23, in init
    modules = get_all_submodules(Path(__file__).parent)
  File "/Users/newuser/Library/Application Support/Blender/2.80/scripts/addons/measureit_arch_v01/auto_load.py", line 51, in get_all_submodules
    return list(iter_submodules(directory, directory.name))
  File "/Users/newuser/Library/Application Support/Blender/2.80/scripts/addons/measureit_arch_v01/auto_load.py", line 55, in iter_submodules
    yield importlib.import_module("." + name, package_name)
  File "/private/var/folders/k6/1x1hxh5d6bs54nmjxb030mr00000gn/T/AppTranslocation/283ADC70-B960-4CC9-8C28-6F0E0F4DEBEC/d/blender.app/Contents/Resources/2.80/python/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/Users/newuser/Library/Application Support/Blender/2.80/scripts/addons/measureit_arch_v01/measureit_arch_annotations.py", line 42, in <module>
    from .measureit_arch_main import get_smart_selected, get_selected_vertex
  File "/Users/newuser/Library/Application Support/Blender/2.80/scripts/addons/measureit_arch_v01/measureit_arch_main.py", line 40, in <module>
    from .measureit_arch_geometry import draw_annotation, draw_alignedDimension, draw_line_group, draw_angleDimension, update_text, draw_vertices, draw_object, draw_edges
  File "/Users/newuser/Library/Application Support/Blender/2.80/scripts/addons/measureit_arch_v01/measureit_arch_geometry.py", line 54, in <module>
    Dashed_Shader_3D.fragment_shader)
Exception: Shader Compile Error, see console for more details

kevancress, 2019-03-25:
Hi Blenderbugg, Thanks for reporting the issue, I'll look into it today.

Looks like its an problem with the OpenGL shaders I've written, Could you let me know what Operating System and Graphics Card you're using, it'll help track down the issue!

kevancress, 2019-03-25:
Found some things that might be causing compatibility issues in the fragment shader . Just committed the changes, give the latest version a shot and let me know if it works!

Blenderbugg, 2019-03-25:
cool, I will.
My system is: 
Operating system: Mac Pro (Mid 2010) macOs 10.13.3
Graphics card: Radeon RX 580

Blenderbugg, 2019-03-25:
It is installed and working!
See how we go. Nothing like Beta code!
I will see if I can break it for you...... OMG

kevancress, 2019-03-25:
That's great! 
Break it as much as you can, thats the best way to make it better 👍

closed 2019-03-25 by kevancress