kevancress/MeasureIt_ARCH #2: Measure fails when trying to enable in Prefs.

garyhaus, 2019-03-25:
[image]

kevancress, 2019-03-25:
Hey Gary, Thanks for the error report. Just made some changes that should fix it.

Try removing the old version in the preferences and installing this one: [MeasureIt-ARCH-master.zip](https://github.com/kevancress/MeasureIt-ARCH/files/3005589/MeasureIt-ARCH-master.zip)

Let me know how it goes!

garyhaus, 2019-03-25:
FLAWLESS now!  Working sir thanks!

kevancress, 2019-03-25:
That's great! Thanks for testing! Let me know if anything else comes up!

garyhaus, 2019-03-25:
[image]
looks fine in edit mode

garyhaus, 2019-03-25:
[image]
broken in object mode?  or is this intended?

kevancress, 2019-03-25:
Intended but I could probably communicate why better. 

In edit mode the annotation gets anchored to the selected vertex, in object mode it gets anchored to the objects origin, I should probably point that out in the tool tip or give a popup warning

garyhaus, 2019-03-25:
ok thanks for the response

kevancress, 2019-03-25:
No problem! Feedback like that really helps me try and make things more intuitive for the next release.

closed 2019-03-25 by kevancress