kevancress/MeasureIt_ARCH #3: "Annotations" missing

Blenderbugg, 2019-03-26:
Hi,
"Annotations" menu item not appearing under "Custom Properties"

[image]

kevancress, 2019-03-26:
Hey Blenderbugg,

Right now the menu panels don't show up unless the active object already has an annotation added to it, same with the dimension and line menu panels.

If its still not showing up after an annotation has been added let me know.

[image]

Blenderbugg, 2019-03-26:
Ah yes... got it. Is there a way to have it on by default? AT least you would know its there to use. Anyway, I am making a little list of suggestions and observations as I go - Where should I post them?

Eric

kevancress, 2019-03-26:
Was hoping to try and reduce UI clutter by hiding them when they're not used, but your right having it be more discover-able would be really valuable! 

In the next couple weeks I'm hoping to rework the settings UI into something similar to [This Modifiers UI Concept](https://twitter.com/FinEskimo/status/1107053329312608257) It's not too cluttered and fits well with the rest of blenders UI system I think?

Thanks so much for putting together a list of suggestions!! I think right now the Issue tracker is the best place to have discussions on these things that keeps them tied to the project and easy to keep track of, maybe I'll start a Features and Suggestions thread to keep everything collected, and I can break out individual projects from there?

closed 2019-03-26 by kevancress