kevancress/MeasureIt_ARCH #4: Features & Suggestions (The place for new ideas & general suggestions)

kevancress, 2019-03-26:
This thread is the place to leave feature suggestions and general feedback for MeasureIt-ARCH.

Blenderbugg, 2019-03-26:
Hi,
Have an idea. Use a pop-over similar to the helper in HardOps. It mirrors important information in the Properties Panel, great for working on the fly or when you properties panel is dragged down to reveal collections. I have almost weaned myself off going to the properties panel while modeling and working in the helper.

[image]

kevancress, 2019-03-26:
Hey,

Great suggestion, would definitely streamline workflow a lot and I shouldn't be too difficult to implement!

kevancress, 2019-03-26:
I've put together a project board [here](https://github.com/kevancress/MeasureIt-ARCH/projects/2) where you can see features I'm currently planning to implement and where I can tag Ideas from this discussion to keep track of them

Blenderbugg, 2019-03-26:
Cool, glad to help you with this. I am a retired Graphic Designer and spend most of my time SCUBA diving and playing in blender. I still do a little graphic design and short movies for web sites plus a little architectural visualization just to keep me fresh.

More suggestions.

1   Leave "show/hide MeasureIt" on for a file you have saved and closed so that when you reopen it you do not have to "reactivate" the addon. I actually thought there was something wrong this morning when I opened the file and there were no dimensions on my drawing till I realised I had to hit "show/Hide"

2   Have some default styles already set. For example Set out plans could have their plan view set to XY for dimensions, while front and back elevation have theirs set to XZ and sides have theirs set to YZ. You can also preset the Flip Text to the correct alignment for those views. Also, I couldn't see "flip text X or Y" available as a choice in styles

3  Have the ability to click on a dimension on the plan and have the corresponding Dimensions tab in the Properties panel highlight or expand. Might make it easier to find things when you have a million (more or less) dimensions to look through.

I have a few more but I really need to do a bit of photoshoping to explain them.

kevancress, 2019-03-26:
These are great!

Really appreciate your help, It's nice to have a set of fresh eyes on this, especially from someone with some experience in architectural visualization!

I've added these onto the projects page, 1 & 2 should be fairly straightforward. 3 might take some time, but is definitely a priority. I'm hoping to add some more interactivity to the 3D view space, click and drag to re-position dimensions, that kind of thing, and combining that with a popover context menu that lets you edit a dimensions settings in place would be really useful. Still wrapping my head around the code for that one though!

Thanks so much for your time putting these suggestions together, I look forward to the next ones!

marcusulpius, 2019-08-19:
Can we have an option for "forced" external arrows on short length lines where the total arrow length approaches x ratio of text?

See image, top left.

[image]

kevancress, 2019-09-22:
Hey Marcus, just got a working start on moving the dimension text, still not too pretty yet but its a start, should be in the latest master (which should work with 2.81 now too, there was a bit of a bug there). Sorry developments been a bit slow, this months been pretty busy.

No donation link for the project yet, I've been working on this as part of my masters thesis for now, but once that's wrapped up In October I might put one together!

monbarcelos, 2019-09-26:
Hello! I have some ideas and questions :+1: 

- There is a way to set the dimmensions like SketchUp layout, using just the vertex snap instead of enter object mode?

[image]

- There is a way to set a inner color for wall to create sections? I saw something like this in yours videos, but you can choose the color?

[image]

- What about a option to set dimmension text size by scale? Like 1:50 or 1:100
[image]

- Is it possible to use it in symbols imported by CAD? 
[image]

kevancress, 2019-09-26:
Hey monbarcelos

Thanks for the Feedback!

1) Dimension creation like SketchUp Layout: No plans for this right now, MeasureIt-ARCH uses the vertex index of the object to keep its position, so changing to a system similar to SketchUp for adding Dimensions would require a pretty big overhaul I think.

2) Cut Color: I'm actually just using Blenders normal Boolean operators to assign cut colors. If the object your cutting has a slot for your 'cut' material, and the object your using to do the Boolean cutting has that material applied as well, then the cut faces will automatically be assigned that material.
[image]
 
3) Scale Dependent Text size: This ones definitely on the road map, I've been working on a [separate addon](https://github.com/kevancress/per_camera_resolution) to help manage multiple cameras all with different sizing and scales. The plan is to merge that functionality into MeasureIt-ARCH and then provide an option to make the text size related to the cameras defined scale.

4) CAD Symbols: I've been thinking of adding a new symbol type that could have a set of common symbols, but for now the best way to achieve with your own blocks/symbols would be to import them as .dxf into blender and use MeasureIt-ARCH line work tools on them.

monbarcelos, 2019-09-27:
Thanks for the tips! Amazing!

Moult, 2019-10-01:
Hey @kevancress ! I'm building BIM/IFC support into Blender - combined with this would be a great workflow for architects! Would you be interested in catching up online so we can demo the two to each other and see if there is space to integrate? My email is dion@thinkmoult.com - I was thinking I could do a quick video demo to show you what it could do, and vice versa, and perhaps we can discuss from there!

https://github.com/IfcOpenShell/IfcOpenShell/tree/v0.6.0/src/ifcblenderexport

I made a short video of it here:

https://peertube.social/videos/watch/3006a205-bfad-46f4-863c-3602cc5a75da

kevancress, 2019-10-03:
Hi @Moult! Just gave the exporter a test, that's fantastic work you've been doing, being able to export to IFC is a huge step for blender and its possible application in OpenBIM workflows!! I think it would be fantastic to collaborate and have a chat on how these projects could integrate and develop further! I'll shoot you an email!

In the meantime I've got some video's of MeasureIt-ARCH up on youtube [here](https://www.youtube.com/watch?v=MWo87QvcEPk&t=).

Elwex123, 2020-02-09:
Thanks for the addon, a much better solution then the basic Measureit addon. 
Few features that would make my life easier using Blender as a flexible CAD program:

A. Gizmos: When there are a lot of measures in a single object, it gets complicated to delete one specific measure because you have to dive into the measurements list. Blender can also suffer from the same issue, but it solves it by allowing users to delete the items they want right from viewport. So a handy gizmo to delete (and hide) measures would be fantastic.

B. Styles: Styles as to right now don't support global measurement distance and offset from target vertices. It would be extremely handy to be able to configure those parameters with the Styles system.

C. Modifiers: Sometimes I want my measurements to only evaluate certain modifiers (For example, evaluating lattice, but not subserf, etc.), but currently I can evaluate the modifiers only in a binary way: all of them, or non of them. It would be helpful to be able to say to the addon to measure until a certain point in the modifier stack, and ignore further changes done to the targeted mesh.

D. Shape keys: Edit mode has a lot more precision modeling power to it them the current modifier system, but we "stuck" with it because it is non destructive. Shape keys bridge that gap by allowing users to save an edit done within edit mode, control it with a simple slide bar, and even to exaggerate the edit. So making the Measureit system support Shapekeys would be a huge jump forward, and allow us to easily present and modify on the fly CAD products to costumers, etc..

E. Performance: For some reason displaying 10 measurements (without ticking the "evaluate modifiers" box) drops the viewport frame rate to lower then when sculpting 1+ million meshes...

F. Bounding box: Sometimes a product consist of a couple of objects, and knowing the bounding box dimensions of the product is necessary. Currently, you can only display the bounding box of objects separately, and that's fine, but it limits me from designing complex and easily manipulated products that consist of a handful of modifier techniques. A feature that would solve this problem is one that would calculate the bounding boxes of all the selected objects, and combine that with the distance between one another, and display that as the bounding box of all selected objects (or something like that...).

G. Instances: I don't know if that's possible, but treating instances as regular objects in the sense of adding bounding box would be nice. It would make duplication of objects more usable to us when using the addon.

Again, I REALLY appreciate your work. It helps us create artistic models and make them emerge to life. Thank you!

kevancress, 2020-02-11:
Hey @Elwex123!

Thanks these are great Suggestions! My comments below:

A) Great Idea! Definitely need to add more in-viewport control through gizmos. For Showing/Hiding and deleting dimensions though I'm thinking gizmo buttons for each dimension might start to make the viewport a little cluttered. My thought instead was to have a better selection system for MeasureIt-ARCH elements. So clicking on a dimension or using it's placement gizmo in the viewport would make it the currently selected dimension in the list UI, so it's easy to find and delete. It might be possible even to add the delete dimension operator to a keyboard shortcut as well? Let me know what you think!

B) I definitely need to include this! I'm thinking of having the distance and offset controlled first from the style, but allowing for "tweaks" or overrides in individual dimensions that would add or subtract from the style to allow for final adjustments.

C) I'd love to add more fine grained support for modifiers, but (as far as i know) I'm limited here by how blenders dependency graph is exposed to the python API. I too only have the binary option to either evaluate the depsgraph (and modifiers) or don't evaluate the depsgraph before drawing the dimensions. Someone who knows Blenders API and source code better than I do might have a clever way around this, but for now I don't know if there's much we can do on this one. Maybe this will change once Blender's modifier system gets redesigned in the everything nodes project??

D) I just did some testing, and it looks like MeasureIt-ARCH dimensions do respond to shape keys!! I'm a bit surprised by this since I haven't intentionally tired to make them work, but it seems to follow the mesh changes without issue! Let me know if any problems come up in your workflow with this though, because it sounds like a great way to work and I'd like to support it!

E) The two major performance issues right now seem to be my draw call batching, and the method i'm using to check the object geometry that the dimension is attached to. The second one is unfortunately dependent on the vertex and face count of the object being dimensioned not the number of dimensions in the scene... I've been testing out some methods to improve both these issues, but they both involve so pretty serious refactoring. Definitely a priority, but might take a while to resolve unfortunately.  

F) Great suggestion, shouldn't be too difficult to allow bounds dimensions to check over multiple object bounding boxes!

G) I think this is possible I need to work through my instancing code to better support the existing dimension types already. I'll be sure to look into adding bounding box support to that too when I do!

Thanks so much for the suggestions! I really depend on Feedback like this to keep improving MeasureIt-ARCH. Really appreciate it!!

Elwex123, 2020-02-13:
Thanks for taking your time to replay to all of my comments and suggestions, appreciate it.

A. Agee with you on that. Selectable measures would be fantastic. It can be expanded to have its own context menu, right? Big UI potential right there.

D. As I tested it further, it seems like it does work perfectly fine, but a bevel modifier I put stopped the evaluation of all mesh deformations, and I accidently thought that as were the result of ShapeKeys compatibility.

If talking UI stuff, it seems quite odd that the measurements text is rendered as a pixelated image, and not "vectorized" one like any other UI element in blender. Is it intentional? Because the original MeasureIt elements are indeed vectorized, the linework and the text itself.

I though of another feature that would be useful for precision modeling where the measuring process needs to tightly integrated with the actual modeling process, like when modeling out of a technical sketch. This feature would be an option to automatically add a measure when extruding between the active vertex (the last selected) and the extruded vertex. Currently, your workflow is constantly interrupted by the manual process of adding measures as you proceed modeling. Because this isn't needed all of the time, a convenient toggle and hotkey inside edit mode would switch this feature on and off.

Hope I'm not asking to much from you, but Blender has huge potential as an artistic CAD software, and your addon can really push the limits here. Thanks for your effort!

Elwex123, 2020-02-14:
Just thought of an idea for measures to not be linked to the actual mesh data, meaning no problems with certain generative modifiers (Mirror with flipped bisect, Boolean etc.). You take an edge and shrink-wrap it to the target object. It can be locked to a specific axis with project mode, make it snap to vertices with nearest vertex, or leave it at nearest surface point method. It can be thought of as a ruler, but on steroids. Of course it has the down side of not following the target mesh if moved around (although it can be solved with parenting), it cant be dragged within object mode, and its a bit tedious to add the shrink-wrap modifier and parent it to the target object, but its has huge, HUGE potential. 
Honestly, I think it can certainly replace the current native measurements (of blender, not Measure ARCH measures), just more UI and UX touch-ups on the accessibility and ease on use (measurements selection etc., maybe an icon at the toolbar?), and we have a new, much better measuring system that would replace the current native blender ruler system.

One more think, and you probably think by now that I'm a spammer, is it possible to make measures text visible even when covered by objects, like the in-Front option to objects found under the viewport display panel? Would be very handy.

Well, that's it for now, if I'll have more ideas, I'll share them.

And again, thanks for making Blender an artistic CAD program!

kevancress, 2020-02-17:
Hey @Elwex123!

Great Stuff, my thoughts below:

A) I think I've got the basics of the selection working (just for axis and aligned dimensions right now but it's a start). I'll drop a GIF in the Gizmo brainstorming thread of it in action, and It's in the master branch now if you want to test it out!

D) Great that the shapekeys are working. That does remind me though, I do stop modifier evaluation when certain generative modifiers like bevel are added to the stack, but I don't really communicate it at all to the user when it happens... I should probably add in an error message or something when an unaccepted modifier is added to make this more clear! (At least until I can work out a better way to handle generative modifiers!)

1) Dimension Text: Yeah I've been struggling with this one a bit. The old MeasureIt used blenders .blf text drawing in screen space. It was still technically a raster image, but since it was always drawn relative to the window size it always came out crisp like vector text (this is also why it stayed the same scale regardless of your zoom, because its size was relative to the viewport window.) To get the text in MeasureIt-ARCH to feel like it's really attached to the dimension in 3D space I needed to draw in the 3D world space... but blenders blf text drawing is broken broken in world space for 2.8+ so I'm using a hack-y work around where I draw the dimension text in screen space off-screen, store that as a 'texture' in the dimensions data, and then uv map that texture to a 3D card on the dimension... the downside is that you can zoom right in on the text and see how pixelated it is. I could use vector text, but it would make it a lot harder to let the user use system fonts... Long story short, my plan right now is to eventually have a proper system for setting the camera scale and paper size (https://github.com/kevancress/per_camera_resolution), and then the resolution can be derived from that. 

2) Dimension on Extrude: I really like this Idea, would make modeling and dimensioning really seamless! Might take me a bit to figure out how to tag my script onto the end of blenders existing operators like extrude, but I feel like it should be do-able.

3) Shrink wrapped Dimension: Really cool, I'll have to test out this workflow and see what I can add to facilitate it. Right now i've been using dimensions attached to empties hooked to verts if i want to avoid a direct connection to the mesh data, but this sounds easier to manage! 

4) Draw in front: Never occurred to me before but it was super to simple to add, it's in the master branch now for dimensions and annotations!

Ps. Don't worry about 'spamming' It's always great to have a nice collection of ideas to work on and mull over for this, so if you've got ideas keep em' coming! I can't guarantee I'll be able to implement everything, and it might take me awhile to respond sometimes but knowing the features people want is always super helpful! 

Cheers!
Kevan

globalcitizen, 2020-06-02:
- [ ] __Add quick 'print preview' button__ to alter the default workspace settings to be more 'drawing-like' such that the blender view is effectively an instant preview of the final output. This would consist of changing the background color for the view to white, disabling the grid and disabling the gizmos, enabling visibility of all annotations/dimensions for the current camera/view.
- [ ] __Add quick 'generate elevations' button__ to generate cameras that encompass all scene elements and represent orthographic projections.
- [ ] __Add a floorplan or cross-section definition feature__ that generates both cameras and object cutaway previews to facilitate easy dimensioning of these common drawing types. Optionally, add an option to name these and generate indicators using the standard symbol like this. [image] It may be worth reviewing the [code in the existing stored views addon](https://github.com/blender/blender-addons/tree/master/space_view3d_stored_views) as a reference.
- [ ] __Integration with solar addon__ to (1) detect north point and include this as a special annotation in the absolute corner of any output or as a gizmo overlay. (2) Correctly label quick elevations with cardinal directions.

Moult, 2020-06-02:
Just adding to the discussion some of the progress made in the BlenderBIM Add-on world that relates to this. But first, a picture - an annotated floorplan, annotated in Blender, but cut directly from IFC, and all annotation geometry stored in IFC:

[image]

The approach taken in the BlenderBIM Add-on has both similarities and differences to MeasureIt-ARCH. The biggest difference is that the annotation in the BlenderBIM Add-on is actually almost Blender agnostic - Blender may be used to create content, but the actual annotation is purely IFC + SVG. Also, all of the annotation geometry is actually stored in specific annotation geometric representation subcontexts in the IFC data itself. All smart annotation data (e.g. tags) actually read from IFC data itself. All hatching styles are derived from IfcMaterial data. The UI aspect to help place these annotative elements, however, is extremely underdeveloped, unlike MeasureIt-ARCH which is much more polished.

More reading [in this thread](https://community.osarch.org/discussion/63/resurrecting-annotation-storage-in-ifc/p1).

I think we should revisit the opportunity to combine the capabilities of both add-ons :) I subscribe to all the events in this repo and the work done here is astounding! Great work @kevancress !

globalcitizen, 2020-06-02:
@moult Hey that looks awesome. I have no idea what IFC is though. I have a vague idea it's some sort of industry standards push in the exchange format / BIM space from closed source world players. All I want is to design in Blender (not Revit) and render from Blender, not use closed source world stuff, but get all of those awesome rendering features you listed. In the demo shown, while the drawing "looks the part" it doesn't guarantee workflow (eg. section labels actually have sections associated in blender) ... and some of the symbology is nonstandard. That said, I will give it a try.

Moult, 2020-06-02:
@globalcitizen IFC is the ISO standard for how to store BIM data. It is run by a non-profit organisation called buildingSMART, and nowadays is contractually required by most industry players and increasingly more governments. Historically, it was heavily influenced by closed-source vendors like Autodesk, mostly because there wasn't much of a FOSS offering in BIM back then, but times have changed. It is supported not only by Blender (through the BlenderBIM Add-on and Archipack), but also by FreeCAD, OpenProject, IfcOpenShell, XBim, EnergyPlus, xeogl, BIMServer, and more. If anything, FOSS software has better IFC support than the closed vendors. There is absolutely no requirement to use any closed source software, and I myself do not use any. You can have a 100% Blender workflow if you wish, which I encourage and also do myself. I only mention Revit because I needed a test model as base geometry to then demonstrate Blender's annotative abilities that could be shared to the public, and that Revit IFC model was publicly available.

A section is represented in Blender as a camera object. Workflow is a high priority, and I have used it myself in small commercial jobs to deliver drawings, and am very aware of the current shortcomings. Symbology can be customised completely - what is shown is based on Australian architectural drafting standards.

kevancress, 2020-06-03:
Woah @Moult  thats fantastic stuff!! I downloaded the source for BlenderBIM last night and was testing out some of the new features you've added, the custom section cut shader is amazing, and the .svg export is fantastic! I'd though that annotation & dimension elements had been totally removed from IFC but it looks like your bringing them back in a big way! It would be amazing if we could synchronize the outputs and display of annotation elements through a combination of MeasureIt-ARCH and BlenderBIM (something like automatically generating MeasureIt-ARCH display elements for IFC annotations for display within Blender, and then using some combination of SVG & IFC for the creation of doc packages and cross platform collaboration).

monbarcelos, 2020-06-03:
@Moult, there is a video showing how to use dimmensions that way?

globalcitizen, 2020-06-03:
@monbarcelos I asked this two days ago. There are no current videos because the pace of development has outstripped Moult's ability to produce them. The UI changes a lot. However, right now the critical different between __MeasureIt-ARCH__ and __BlenderBIM__ in terms of dimensioning is this: 
- __Measureit-Arch__ allows __real time dimensioning__ where the values are dynamic and associated with the current state of the object, so that any object transform will automatically update the dimension. However, its output is __bitmap only__.
- __BlenderBIM__ still has __one time dimensioning__ where the values are auto-generated but remain static and do not update with the current state of the object, so that __object transforms require manual re-creation of an export__ which will trigger it to use updated dimensions. Its __output is SVG with embedded bitmap and partial vector elements (including the dimension lines, symbols and text)__.

smoberlin, 2020-07-17:
Using MeasurIt-ARCH to dimension plots of land for subdivision and construction projects, so measurements are 10s to hundreds of feet.

Currently forced to make fonts ~200 or larger for them to be large enough to be seen on screen and they do not scale with view window zoom. Nice enhancement would use screen space to size dimension scale in style settings so they are zoom-independent. Same for area.

Also, being able to set fixed distance/offset for dimensions in style would be very useful. Often want to set to zero when marking boundaries of land parcels on a topological map or subdivision map, and there can be many measurements so it's painful to have to set it manually in each dimension.

Because measurements are large distances (imperial), precision to inches is too fine. It would be great to be able to limit dimension precision to feet-only (round inches to nearest foot).  

Thanks for an outstanding extension of MeaureIt!  Looking forward to the continued enhancements, etc., discussed by others, too.  :)

Moult, 2020-07-27:
Can the default colour of the annotation be white, or something which complies with WCAG contrast requirements? https://contrastchecker.com/

Right now, black on the Blender default grey is a bit hard to read :)

meikasahara, 2020-07-27:
The implementation of theText Autoplacement toggle works great! I noticed that when it's enabled, the arrows indicating the measured dimension never flip outwards though when the text alignment is set to center. This brings me to suggest that an extra toggle to determine when arrows should be in the interior or on the exterior of the dimension out would be a welcome addition! Thanks!

globalcitizen, 2020-07-28:
Here are some random ideas to potentially solve @Moult's suggestion of colour contrast enhancement:
- Migrate default color to a 'difference' filter, ie. render the text color as the exact inverse of the color below.
- Render text with a drop shadow in the inverse color, ie. if text is white, surround it with a few pixels of black, and vice versa. This can be achieved with centered placement and an initial rendering of the text in the other color with a slightly larger font size. Bonus points if this drop shadow size is configurable.
- Render text on a solid shaded panel of inverse color

Here are some suggestions for @smoberlin's concern:
- Have a checkbox which, when enabled, applies dynamic scale within a normal editing context to ensure legibility.
- Consider the potential to associate specific scale factor configurations with camera views.

globalcitizen, 2020-08-07:
Either:

- More frequent releases (like after every fix); or
- Cross-platform documentation on how to run the git version.

As the former is easier, I recommend the former.

Moult, 2020-08-07:
@globalcitizen from my understanding to run the Git version you kinda just pull the latest and put it in your add-ons dir verbatim. There is no packaging required :) At least I've been doing that and it hasn't borked yet.

globalcitizen, 2020-08-07:
Mine borks as per a previously supplied screenshot. I can retry and give
you a new one today. I am usually on OSX but will also test on Windows.

globalcitizen, 2020-08-07:
Confirmed working on Windows. Latest revision and Blender 2.83.4.

Although:

* determining the subdir is a bit hard for regular users
* noticed Github default subdir name is different to release zipfile
enclosed subdir name (IIRC).. dunno if that means a subsequent zip file
release install may cause ugly clashes with the git one?

OSX to follow.

Moult, 2020-08-07:
@globalcitizen the Github subdir name needed to be changed so that it could be imported by other add-ons. Originally, from my understanding, `MeasureIt-ARCH` was released by itself, with no issues. However, now that Archipack and the BlenderBIM Add-on are attempting to integrate, we need to import the Python modules, and "-" is not allowed, therefore the name was changed to use an underscore relatively recently in the releases.

globalcitizen, 2020-08-07:
Sounds like a repo rename or `git mv` may be required. Checking OSX now.

globalcitizen, 2020-08-07:
Confirmed working on OSX. No idea when that changed.

For now, I will add a summary of the process for checking out on each platform to the docs and push it.

Longer term this is a crap install process, however, and should be fixed as suggested.

globalcitizen, 2020-08-07:
Apparently unable to fork or edit Wiki documentation at https://github.com/kevancress/MeasureIt-ARCH/wiki/Version-0.4-Documentation ... therefore suggest rolling this in to `README.md` as is more common in open source projects (in my experience). I will do that and push the result. Side effect: reduces reliance on and lockin to Github.

kevancress, 2020-08-10:
Hey @Moult @globalcitizen 

I've just changed the repo name too which should help with the install issue

globalcitizen, 2020-08-12:
Cool. New suggestion: "Clear all dimensions" button with a suitably scary confirmation popup. Quicker than clicking one by one to clear old work.

meikasahara, 2020-08-26:
**Proposal: Vectors instead of floats for Offset and Rotation in Dimensions settings.**
Using 3d vectors and euler rotations in Dimension settings could perhaps enable the user to offset and rotate the measurement more freely, when Auto-Placement is disabled

Shaddy-ElDesouky, 2020-11-23:
Hello @kevancress @Moult I was trying to figure out how to make all my workflow(Set design-Interior-Arch) inside blender and I discovered BlenderBim and Measureit-Arch seperatly, but after I read this page and the merging suggestion I'm so happy, and understood that I'm on the right track.

Please try to push this as fast as you can, unfortunately I can't code but I could help by bug reporting and suggestions, also how to follow the news,updates? is there a time line? Have you started already?
shaddyeldesouky@gmail.com, blender@shaddydesign.com

baswein, 2021-06-04:
Thanks for the great add-on! 
I am trying to label some elevations using the annotation tool that get the value from the global vertex location. And it doesn't seem possible. I tried to set up a custom property with the driver but couldn't get it to work (it's all pretty new to me). But it seems like you have the location of the vertex so it shouldn't be too difficult? Unless it's a Global, Local, Datum, Curvature of the earth kind of nightmare :) Anyway thanks again!
[image]

kevancress, 2021-06-05:
Hey @baswein 

That's a great suggestion! I should definitely expose the vertex location in world space as an annotation label option, I can probably add that in sometime next week!

In the meantime though, here's how I've been approaching spot elevations when I've needed them:

I generally use one of Blender's empty objects, and attach the annotation to it, then I set the annotation text field to "Custom Property" and enter `.location[2]` in the text field to make the annotation display the Z height of the empty in world space. Then I turn on snapping and set the "snap to" to face instead of increment. Then I drag the empty to the location where I want to take the spot elevation, and it should snap to the mesh surface.

Here's a GIF of the setup:

[image]

This isn't really the most user friendly / discoverable setup though, so I'll definitely add an elevation option for text fields ASAP.

Cheers,
Kevan

baswein, 2021-06-05:
Perfect! 
That works well. Thanks! Do you mind if I post the work around on OSArch?
One issue I see is that in the US we typically use engineering units for elevations. Like this: 1.25'. but often use architectural units for other measurements. If anyone would like to sign a petition to make the US metric it is here https://sign.moveon.org/petitions/change-to-the-metric : )

Condiolov, 2021-06-29:
I love this plugin just perfect! I'm even testing in blender 3.0 this perfect so far I would like to suggest that when quoting the measures their provisional name should be the measure, and when you click on the measure it changes the color, for example it turns orange so it would make identification easier, by the plugin!!

Myrkskog, 2025-03-24:
@kevancress I realise this is a user dependent problem. I tend to force evaluate depsgraph sometimes per object (reasons). If the file has a lot of collections, with many objects, it is hard to manage which still have this active.

Might you consider a kill-all / global reset for evaluate depsgraph?

It is a slog to manually check every object.

Maybe I should just use evaluate depsgraph on a scene level and toggle it off right after the update.