kevancress/MeasureIt_ARCH #6: Line weight not working

Blenderbugg, 2019-03-26:
In Object Mode - Line Weight - Edges not increasing only verts are showing as bigger. 
[image]
[image]

kevancress, 2019-03-26:
Did a little research, this looks like another difference between how Windows and MacOS support OpenGL line widths for drawing... look like the Blender devs have put some work into moving away from using OpenGL for this in the [source code](https://developer.blender.org/D3502?id=11321) I'll see if i can access these new draw calls from the add-on

kevancress, 2019-03-26:
Last commit should fix this. I'm handling the line thickness with a geometry shader rather than using glLineWidth. The downside is that I still need to manually implement Anti-Aliasing in the shader, so for now lines will be a bit jagged. Let me know how the new version looks, there might still be some issues around corners

kevancress, 2019-03-26:
Just brought back the Anti-Aliasing support so the jaggies should be fixed.

Blenderbugg, 2019-03-27:
Looking good! The line weight numbers seem to make more sense. You don't seem to need to set a Z offset either.

closed 2019-03-27 by kevancress