kevancress/MeasureIt_ARCH #7: Measurements "blacked out" on opening file

Blenderbugg, 2019-03-26:
Opened my file, measurements appearing as black blocks. Click on colour tab in Dimensions box, numbers return.
[image]
[image]

kevancress, 2019-03-26:
I've been able to reproduce this one on windows too with files with large numbers of dimensions. Strange that all the dimensions are returning the same value when the text comes back.  For me they come back to their correct values if the geometry is modified, is that the case for you?

It looks like the method I'm using to draw text has some hiccups. I'm having to work around some strange behavior of the text api in python in 2.8 might have to do some digging in the source to find a full fix to this one.

Blenderbugg, 2019-03-26:
I have made a little video and put it up on a workspace I have for this sort of thing.....

https://youtu.be/k6t2MUdWzzM

kevancress, 2019-03-27:
Wow! This one was a fun one, thanks for the video, that really got me on the right track to get this sorted out. Turns out the memory locations that I was assigning to store the text for the dimensions could get randomly reassigned when reloading the file and would overlap with each other, and occasionally with memory locations storing bits of the user interface.

Should be all fixed in the last commit, it seems to be working on my end at least, let me know if it fixed the problem for you too

closed 2019-03-28 by kevancress