kevancress/MeasureIt_ARCH #9: Addon not loading MacOx

Blenderbugg, 2019-03-27:
March 26 and 27th Blender 2.8 builds for MacOx not loading MeasureIt or MeasureIt-Arch.
When I deselect and re-select the addons in preferences following error occurs.

bpy.ops.mesh.primitive_cube_add(size=2, view_align=False, enter_editmode=False, location=(0, 0, 0))
Activated BoxCutter
bpy.ops.object.editmode_toggle()
bpy.ops.mesh.select_mode(use_extend=False, use_expand=False, type='VERT')
bpy.ops.machin3.vertex_mode()
Traceback (most recent call last):
  File "/private/var/folders/k6/1x1hxh5d6bs54nmjxb030mr00000gn/T/AppTranslocation/761CA1A9-DA18-4F2D-B7D2-024B2255EF2A/d/blender.app/Contents/Resources/2.80/scripts/modules/addon_utils.py", line 384, in enable
    mod.register()
  File "/private/var/folders/k6/1x1hxh5d6bs54nmjxb030mr00000gn/T/AppTranslocation/761CA1A9-DA18-4F2D-B7D2-024B2255EF2A/d/blender.app/Contents/Resources/2.80/scripts/addons/measureit/__init__.py", line 150, in register
    register_class(cls)
ValueError: register_class(...): already registered as a subclass

Traceback (most recent call last):
  File "/private/var/folders/k6/1x1hxh5d6bs54nmjxb030mr00000gn/T/AppTranslocation/761CA1A9-DA18-4F2D-B7D2-024B2255EF2A/d/blender.app/Contents/Resources/2.80/scripts/modules/addon_utils.py", line 384, in enable
    mod.register()
  File "/private/var/folders/k6/1x1hxh5d6bs54nmjxb030mr00000gn/T/AppTranslocation/761CA1A9-DA18-4F2D-B7D2-024B2255EF2A/d/blender.app/Contents/Resources/2.80/scripts/addons/measureit/__init__.py", line 150, in register
    register_class(cls)
ValueError: register_class(...): already registered as a subclass

bpy.ops.object.editmode_toggle()
Traceback (most recent call last):
  File "/private/var/folders/k6/1x1hxh5d6bs54nmjxb030mr00000gn/T/AppTranslocation/761CA1A9-DA18-4F2D-B7D2-024B2255EF2A/d/blender.app/Contents/Resources/2.80/scripts/modules/addon_utils.py", line 351, in enable
    mod = __import__(module_name)
  File "/Users/newuser/Library/Application Support/Blender/2.80/scripts/addons/MeasureIt-ARCH-master/__init__.py", line 94, in <module>
    measureit_arch_styles.MeasureitArchDimensionSettingsPanel,
NameError: name 'measureit_arch_styles' is not defined

Blenderbugg, 2019-03-27:
I have logged the MeasureIt error with Blender Dev https://developer.blender.org/T63027.

I am also having the same issue with Box Cutter, HardOps, and a couple of others. Jerry said sent me the following in an email when I reported the error in his addons to him.

"there could also be something configuration related. \2.80
in the 2.80 area is a config folder.(not sure about mac) but sometimes I will zip it then delete it so blender opens as new to test. "

kevancress, 2019-03-27:
Hmm, just downloaded the latest build, not able to reproduce this on windows, but I've seen similar errors when there's a conflict between two different versions of an add-on that blender's trying to load. I'd say Jerry's suggestions is probably a good one but you probably just need to clear out the particular add-ons causing the problem and re-install them from scratch.

Not too sure how blender sets up the config folder on Mac but from the error messages it looks like deleting the addons from "/Users/newuser/Library/Application Support/Blender/2.80/scripts/addons/" and then re-installing might solve the issue, that's my best guess anyhow.

Blenderbugg, 2019-03-27:
Hi ya,
Tried that, no joy. Blender and Macs! Seen as its happening with Measureit at well I would say its Blender problem. I have logged it with them, see what happens you may not need to do anything.

Eric

kevancress, 2019-03-27:
That's tough, Well I'll keep an eye out to see if there's anything I can do on my end

Blenderbugg, 2019-03-28:
cool. At the moment I am back on the build from the 24th where everything is fine.

Blenderbugg, 2019-03-28:
Good news,
Just downloaded the latest build for MacOs (March 28) and everything is back and working fine. I don't know what made everything fall over for the last three daily releases but it seems to have fixed itself..... for now. My error report is still open on Blender.org so I don't think they have actually fixed it, but nice to know it is not your addon.

kevancress, 2019-03-28:
That's great, glad to hear it!

closed 2019-03-28 by kevancress