kevancress/MeasureIt_ARCH #14: Brainstorming for Interactive Dimensions in the 3D Viewport

kevancress, 2019-05-09:


kevancress, 2019-05-09:
Gizmo's are definitely the simplest way to go here.

Alignment Positioning and Draw shape might be a bit of a trick but it's a solid start

[image]

kevancress, 2019-05-11:
Gizmo's now spawn for all of a selected objects dimensions & are positioned at the dimensions midpoint!

Still need to get them aligned with the proper rotation.

[image]

kevancress, 2020-02-11:
Note: Dimensions should become "active" in the list UI after being manipulated by a gizmo

Elwex123, 2020-02-14:
Hmmmm… Is it possible for the gizmos to follow the measure movement when trying to be aligned to the user's view? I believe it would be pretty hard... And what about a smoother transition when the measure rotate? Is it a limitation, or an intentional decision and I'm sort of missing something? Maybe add another colored arrow at the opposite side of the measure for offset? 
Gizmos for rotation? I personally don't use it quite a lot as the addon does the rotation for me, but maybe a 20ish degree arc near the measure for the gizmo design? Would be pretty interesting...

kevancress, 2020-02-17:
[image]

Here's the current Gizmo Selection implementation!

I think getting the gizmo to match, the measure movement better should be possible, just need to work on when the gizmo refresh is called in the code. 

The transition right now is a bit of a limitation, the dimensions position gets calculated every draw frame so there's not much option to interpolate or smooth out the change. would be nice though!

I'm hoping to have gizmos for leader offset and line thickness too I think!

Elwex123, 2020-05-08:
Hi, @kevancress! 
When working with collection bounding boxes, I find it hard and quite confusing to locate the object which stores the bounding box settings. In the end I have to cycle through all of the objects in the collection in order to find it. I believe that making Measureit ARCH elements selectable (or having a "select button" gizmo) would fix the problem, but it would also need to highlight the object which the selected measurement is assigned to.
Don't actually know how hard it is to code, so I hope I'm not asking for too much. Anyway, thanks for the great addon, it really helps me in my design process!