kevancress/MeasureIt_ARCH #19: Arrow style; length vs width ratio

marcusulpius, 2019-08-02:
Awesome tool!

Is there a way to change the arrow head style on the dimensions lines so they are longer, more streamlined?

Traditionally CAD arrows are longer and wider than the addon exhibits? Arrowheads more typically have  a 3:1 ratio in length vs width. In essence, what I am looking for is more akin to the "closed filled" arrowhead from AutoCAD. Currently, the arrow is too fat.

Love to ideas on how.

kevancress, 2019-08-07:
Hey Marcus,

Thanks for the suggestion! I'v added an Arrow Angle property to the dimensions that lets you adjust the arrows shape. Let me know if that lets you get the look your going for!

[image]

marcusulpius, 2019-08-07:
Wow, thanks!!!

[arrow angle]<https://user-images.githubusercontent.com/16860574/62595551-c9bcf380-b8ac-11e9-88ea-3b352f46338f.gif>

kevancress, 2019-08-07:
Great! If you download the latest master it should be in there now

marcusulpius, 2019-08-07:
Thank you so much!!!

closed 2019-08-07 by marcusulpius