kevancress/MeasureIt_ARCH #22: Permit Arrow Size Below 6; use decimal or fractional for small models

marcusulpius, 2019-08-13:
I cannot change the size of the arrows below a unit value of 6(?). This looks silly next to smaller objects(?).

This means on a cube that is 10", the arrows are laughably too large for the dimensions and object.

Can the smallest arrow size be changed to use decimal or fractions of units rather than hard coded at 6 minimum or even match 1/2 line height scale?

[image]

kevancress, 2019-08-13:
Hey Marcus,

Just pushed an Update that removes the lower limit on the arrow size. It's still capped at one, which works alright down to about 2", I'll try and get a change in soon to use decimal units bellow this... or maybe it's better to automate the arrow size based on the distance measured and then just use the arrow size value to adjust? let me know what you think.

I've also made the extension lines adjust based on the arrow size, noticed they look a little long at small sizes too. 

The new master also has some changes to the automatic dimension placement for aligned dimensions, curious to see how that's working for you, really appreciate the feedback!

marcusulpius, 2019-08-13:
Kevan, you're amazing! I'll try this out later today for sure, and thanks so much for the quick response and hard work!

marcusulpius, 2019-08-13:
I can shrink down to 1 which is great but I no longer see the arrow angle to change the proportion of the arrow head. Issue #19.

kevancress, 2019-08-13:
Hmmm... I can't seem to reproduce that on my end, are you getting any error
messages in the system console? (You can open the Console under Window ->
Toggle System Console)

I'll take a look into it more this evening and see if I can get it working
again!

marcusulpius, 2019-08-13:
Looks good! I'd love to have 1/2's between them for smaller items but this is great now. Thank you.
[image]
[image]

closed 2019-08-14 by marcusulpius