kevancress/MeasureIt_ARCH #26: Error if I Delete a Dimensioned Object

Clockmender, 2019-10-31:
Hello, I have just started working with your add-on and find two things:

1) I can't find an option to Delete a dimension, which may have been placed in error.

2) When I delete the Object with dimensions, I get this error:

```
Traceback (most recent call last):
  File "/Users/mac/Library/Application Support/Blender/2.80/scripts/addons/MeasureIt-ARCH-master/measureit_arch_gizmos.py", line 118, in poll
    if 'DimensionGenerator' in ob:
TypeError: argument of type 'NoneType' is not iterable

location: <unknown location>:-1

location: <unknown location>:-1
```

Can you look into this for me please?

Clockmender, 2019-10-31:
[image]

I found them!, but they are "Read From The Left" not "Read From The Right" when "flipped":

[image]

Can we have this as an option please?

Clockmender, 2019-10-31:
One more thing, when you "flip" a dimension, the dimension lines also flip, meaning they don't leave a gap between object & dimension:

[image]

This one was flipped to ge the text the right way round, then moved with the gizmo.

kevancress, 2019-11-04:
Hey  @Clockmender, thanks for the error report!

Just committed some fixes that should take care of these issues. Give the master branch a shot and let me know if works for you!

1) Added a check to resolve the Gizmos NoneType error when deleting a dimension
2) Dimension text should automatically re-orient itself to remain properly legible based on the users view direction
3) The annotation warning was showing up due to some experimental legacy code that I've removed.
4) I've removed the dimension flip toggle since the text should automatically re-orient now, and updated the draw code so that the extension lines and gap should be maintained as the dimension is dragged around with the gizmo

Hope that fixes it, let me know if anything else comes up!

Ps. Just found your Precision Drawing Tools add-on, looks fantastic!

Clockmender, 2019-11-04:
Thanks!! I am on holiday in Chile at the moment then going to Antartica, I will try to check this when I next get WiFi. Sorry for short response, just off to catch a plane...

If you have some time to check out PDT that would be nice, there is a wiki to help you. (Enough trying to type on my phone)

Clockmender, 2019-11-08:
Hi Kevan,

This looks good! I have not checked everything as I have not had much spare time! There is one question please; is it possible to make the arrows more slender? When I was taught technical drawing, many, many years ago, we would make arrows between 2 & 3 times longer than wide. So if an arrow was 6mm long, it would be between 2 & 3mm wide at the widest point. I don’t know if this is a difficult thing to do, but it would make the dimensions look more professional and more in line with standards, you can see what I mean on the drawing I posted earlier, these arrows look about 3:1 ratio.

Thanks for all your work here!

Clockmender, 2020-01-11:
Sorry to be a pain, but I get this in my console window:

```apple@Apples-MacBook-Pro MacOS % ./Blender
Read prefs: /Users/apple/Library/Application Support/Blender/2.81/config/userpref.blend
found bundled python: /Applications/Blender.app/Contents/Resources/2.81/python
measureit_arch: Imported multifiles
Warning: class MeasureitArchIndex contains a property which should be an annotation!
/Users/apple/Library/Application Support/Blender/2.81/scripts/addons/Measure-it-arch/measureit_arch_dimensions.py:40
    assign as a type annotation: MeasureitArchIndex.glidx
Warning: class MeasureitArchFaces contains a property which should be an annotation!
/Users/apple/Library/Application Support/Blender/2.81/scripts/addons/Measure-it-arch/measureit_arch_dimensions.py:51
    assign as a type annotation: MeasureitArchFaces.glface
register_class(...):
Warning: 'MeasureitArchRenderPanel' doesn't contain '_PT_' with prefix & suffix
Warning: class MEASUREIT_ARCH_OT_rendersegmentbutton contains a property which should be an annotation!
/Users/apple/Library/Application Support/Blender/2.81/scripts/addons/Measure-it-arch/auto_load.py:28
    assign as a type annotation: MEASUREIT_ARCH_OT_rendersegmentbutton.tag
Warning: class MEASUREIT_ARCH_OT_addtolinegroup contains a property which should be an annotation!
/Users/apple/Library/Application Support/Blender/2.81/scripts/addons/Measure-it-arch/auto_load.py:28
    assign as a type annotation: MEASUREIT_ARCH_OT_addtolinegroup.tag
Warning: class MEASUREIT_ARCH_OT_addlinebyproperty contains a property which should be an annotation!
/Users/apple/Library/Application Support/Blender/2.81/scripts/addons/Measure-it-arch/auto_load.py:28
    assign as a type annotation: MEASUREIT_ARCH_OT_addlinebyproperty.tag
    assign as a type annotation: MEASUREIT_ARCH_OT_addlinebyproperty.calledFromGroup
Warning: class MEASUREIT_ARCH_OT_addsegmentortobutton contains a property which should be an annotation!
/Users/apple/Library/Application Support/Blender/2.81/scripts/addons/Measure-it-arch/auto_load.py:28
    assign as a type annotation: MEASUREIT_ARCH_OT_addsegmentortobutton.tag
Warning: class MEASUREIT_ARCH_OT_deletepropbutton contains a property which should be an annotation!
/Users/apple/Library/Application Support/Blender/2.81/scripts/addons/Measure-it-arch/auto_load.py:28
    assign as a type annotation: MEASUREIT_ARCH_OT_deletepropbutton.tag
    assign as a type annotation: MEASUREIT_ARCH_OT_deletepropbutton.item_type
    assign as a type annotation: MEASUREIT_ARCH_OT_deletepropbutton.is_style
Warning: class MEASUREIT_ARCH_OT_listdeletepropbutton contains a property which should be an annotation!
/Users/apple/Library/Application Support/Blender/2.81/scripts/addons/Measure-it-arch/auto_load.py:28
    assign as a type annotation: MEASUREIT_ARCH_OT_listdeletepropbutton.tag
Warning: class MEASUREIT_ARCH_OT_removefromlinegroup contains a property which should be an annotation!
/Users/apple/Library/Application Support/Blender/2.81/scripts/addons/Measure-it-arch/auto_load.py:28
    assign as a type annotation: MEASUREIT_ARCH_OT_removefromlinegroup.tag
Warning: class MEASUREIT_ARCH_OT_addnotebutton contains a property which should be an annotation!
/Users/apple/Library/Application Support/Blender/2.81/scripts/addons/Measure-it-arch/auto_load.py:28
    assign as a type annotation: MEASUREIT_ARCH_OT_addnotebutton.tag
```

This is using the latest master branch files...

kevancress, 2020-01-20:
Hey @Clockmender,

Sorry about the slow reply, been away from development for awhile! Thanks for the heads up, I just pushed a commit that should have fixed the warnings!

Let me know if anything else comes up!
Cheers,
Kevan

closed 2020-01-22 by kevancress