kevancress/MeasureIt_ARCH #27: Can MeasureIt display a furniture design's bounding box dimensions?

redfrog682, 2019-12-27:
I import wrl files of furniture designs to Blender.  These designs have many different parts.  I would like to display overall dimensions like shown in the photo below:

[image]

I have tried to join all the parts, get a bounding box for the overall furniture design, then enable "Edge Length" in Viewport overlays, but all the individual parts' dimensions get displayed.  I end up with a mess like this:

[image]

It seems all parts' dimensions are shown except for the overall bounding box's dimensions, which are what I need.

Can MeasureIt display bounding box dimensions?  That seems to be the most accurate way to display overall dimensions.  Thank you!

kevancress, 2020-02-01:
Hey @redfrog682 

Thanks for the suggestion! I've just started implementing bounding box dimensions. They're still a bit rough right now, the orientation isn't always the best, but it's a start. They're in the master branch now if you want to test them out! I'll be improving them over the new few weeks too I hope.

Cheers!

kevancress, 2020-02-01:
Here's a quick demo too.

[image]

kevancress, 2020-02-28:
Implementation started in 1c15ac486471910d4dab00932e70998fa7249580

See https://github.com/kevancress/MeasureIt-ARCH/issues/35 for future development and discussion

closed 2020-02-28 by kevancress