kevancress/MeasureIt_ARCH #30: Dimensions get [REDACTED]

meutzitzu, 2020-02-04:
when I have no objects selected in the scene the text gets [REDACTED], a.k.a  [DATA EXPUNGED], a.k.a. ██████, and since nobody got this refference i'm just gonna show you a screenshot 
[https://photos.app.goo.gl/sR9cwdw9DFec4RXt6](url)

info that may help : I have multiple scenes in my file, some objects may share names
also, I sometimes used Alt+D to duplicate, meaning some objects share mesh data

kevancress, 2020-02-04:
Hey Meutzitzu,

Seen this issue or variations of it pop up a couple times. Seems to be an issue with my text drawing system, and usually happens more often on mac's (not sure why yet). Here's a couple possible solutions / questions:

1) Try downloading and installing the latest master branch of Measureit-ARCH, I've made some changes to the text drawing system since the last official release that might fix this hopefully. 

2) If it's still an issue, can you open the console (window -> toggle system console) and let me know if there are any error messages showing up there?

3) Are you sure these cabinets aren't part of any top secret project. You may not have clearance to view the MeasureIt-ARCH dimensions for this project. (Got a good laugh out of see a big [REDACTED] in my inbox this afternoon :) )

Seriously though if the first two don't fix the problem or show any errors then if your able to send over part of the .blend file (or just one that reproduces the issue) I can do some more digging and
see if I can get this sorted out.

Thanks for the error report!
Cheers!
Kevan

meutzitzu, 2020-02-10:
upon reloading the .blend it went away 

maybe it has something to do with the fact that Blender can sometimes have troubles recovering from sleep mode, (I've seen my fair share of magenta textures after closing and repoening the lid on my laptop) , so I guess that's that

closed 2020-02-10 by meutzitzu