kevancress/MeasureIt_ARCH #32: Revise default settings & Drawing styles to match ISO 129-1 more closely

kevancress, 2020-02-04:
ISO 129-1 Technical Drawings - Indication of dimensions & tolerances outlines standard dimension drawing guidelines in 2D, should match these as closely as possible when I can

kevancress, 2020-02-11:
cbc36b73788d30cc236d0f4b47de177cb576676d  adjust default arrowheads according to section 5.3.2

meutzitzu, 2020-04-19:
The ISO standard seems a bit too tedious to edit. I get that the options should be there for the more perfectionist folk, but you can never have that degree of flexibillity in a more "automated" system.
what do I mean by "automated system" ?  I would like to bring to your attention how a cute and tiny lil program called SolveSpace handles this https://files.whitequark.org/SolveSpace/solvespace.html

In short, it ditches the alignment of the text to the lines, and it doesn't bother with trying to place the text above or below the dimension line (which could always present problems when stuff is too close together), rather: it places the text smack-dab through the dimension line and just trims out the overlapping part. having used both FreeCAD's more traditional sketcher which works fine in 2D and respects the standards more closely, (i mean you can create real technical drawings with it) I can't help but point out that SolveSpace's more simple and consistent style is better when sketching in 3D space. To be fair: this is a compromising solution, yes and it won't be good for someone trying to achieve perfect, well thought out technical drawings, but rather than having to flip X,Y till I go insane, this would be a nice,quick-n-dirty way to make dimensions more easily readable within the 3d enviornment ... you know .. for those times you want a quick refference for like a 3d printed part and don't want to mess about setting the projection directions beforehand, carefully tweaking the text so it's not upside down, etc, etc.

"But m8, doesn't the regular MeasureIt" already do this ? well kinda ... it's true that the regular measureit  has screenspace oriented text, but it makes no effort to make it easy to manipulate around the viewport (you still have to tweak parameters) and it also doesn't split the line before inserting the text which results in horrible intersections 

i'm not saying this is a much needed feature, just pointing out something which i think could be resonably doable while giving the possibility of working a lot faster