kevancress/MeasureIt_ARCH #43: No MeasureIt Arch not visible in object mode - only in edit mode

MarcelQ, 2020-03-09:
Hi Kevin, Thx for your great work on this add-on. Much appreciated.

I have been trying it out but I am experiencing and issue where i cannot get any of the MeasureIt Arch visuals to show if I am in object mode. If I am in edit mode they display. I have enabled Show in the MeasureIt panel with the option to display for all objects.

I thought what might be the issue is my blender version. I'm using it in both 2.83 alpha and the stable 2.82 but both have the same issue.

I have not tried it in 2.80. However I have noticed in your wiki page you are directing to the daily builds page so that also makes me point out that your add-on doesn't specify what version of blender 2.8x you support. So I presume when you specify 2.8 in your documentation you mean for all 2.80 versions.

My only last resort is to reset my Blender install to defaults and see if that fixes the issue. I have also attached a zipped video for your reference.

Any help will be appreciated.
[Blender 2020-03-09 11-46-51.zip](https://github.com/kevancress/MeasureIt-ARCH/files/4305269/Blender.2020-03-09.11-46-51.zip)

Thx

kevancress, 2020-03-09:
Hey Marcel, thanks for the Issue report!

A couple of quick questions:
1) if you download the [latest MeasureIt-ARCH master](https://github.com/kevancress/MeasureIt-ARCH/archive/master.zip) does that fix the issue (quick note though, the latest master does break styles and linework made with older versions of MeasureIt-ARCH and the application template hasn't been updated yet).
2) If you go to window  -> toggle system console are there any error messages appearing there?

Hopefully the latest master fixes the issue, if not let me know if there are any error messages coming up for you and I'll try and get them fixed. I've had this issue once or twice when the data for a MeasureIt-ARCH elements gets out of sync with the object it's attached to (in the case of your video this is likely because I haven't kept the application template up to date with the code base).

Also you're right I haven't specified a specific blender version anywhere in the documentation, because I'm trying to keep things compatible with any version 2.8 and greater, but I should probably make this more clear in the docs!

Cheers,
Kevan!

MarcelQ, 2020-03-11:
Hi, thx for your follow up and help.

I removed the version I had installed as well as the application template and installed the latest master. Doing this solved the issue of the annotations not displaying in object mode.

The issue was probably a combination of the of the 2 above. I might have another bug or suggestion to report but i will start that in a new issue.

closed 2020-03-11 by MarcelQ