kevancress/MeasureIt_ARCH #45: Could dimensions lines and annotations benifit from a z offset property?

MarcelQ, 2020-03-11:
I have attached an example of where dimension lines have a z fighting issue with 3D objects. This is expected when the lines overlay objects, but sometimes it is necessary to offset dimensions in the direction of the objects. Could this issue be resolved by including a z offset property for dimensions and annotations?

It seems like text doesn't have this issue but perhaps annotation lines will also have an issue like the dimension lines. I have also tested with line styles applied to the objects and although the result is better z fighting is still an issue even when trying different z offsets on the lines styles.

I see the z offset property is marked as a TEMP in the tool tip so maybe you have a better solution that will fix all the issues with z fighting.
[image]

[image]

MarcelQ, 2020-03-11:
Ok I just realized the new draw in front property does solve this issue. I initially thought it was to just draw objects in front of others and not related to the z fighting issue with the same object.

I also didn't see this property initial when testing because i didn't use styles but rather applied the settings per object. When not using styles the draw in front option is not available.

Is my question still valid? Could dimensions and annotations still have z offset? I imagine the draw in front property is using the same underlying functionality (z buffer perhaps?)

kevancress, 2020-03-12:
Good catch! Since I don't think dimensions should need the full range of z-offset I've just hard coded a slight offset to fix the z fighting, that doesn't force the dimensions or annotations to draw in-front of everything (same way that angle dimensions and text avoided z-fighting previously). Let me know if you still think a full slider property for the z-offset would be useful in your workflow though, wouldn't be too much work to add it in.

Also I've adjusted where the 'Draw In Front' property shows up. I should always be at the bottom of the dimension, annotation or line groups settings, or in the style if a style is being used.

Cheers!

closed 2020-03-19 by kevancress