kevancress/MeasureIt_ARCH #57: Issues enabling v 0.4.0

santannamv, 2020-04-18:
Hi Kevan, I am trying to enable the add-on but I keeps showing the errors related on the picture below. I don't know what I am missing here. Thanks.

[image]

kevancress, 2020-04-18:
Hey, Looks to be an issue with the custom shader's I've written for the add-on

A couple things to try:

1) I just pushed a commit with some changes to the shader's this afternoon so it might fix the issue if you download the latest master from today

2) I think there have been some changes to how blender is handling custom shaders recently, so I'd recommend using the stable version of blender 2.82 for MeasureIt-ARCH.

If neither of those things fixes the issue could you let me know the operating system your running, in the past we've had some issues with the custom shaders running on macOS since they don't support some of the same OpenGL functions that windows and linux do... but I might be able to find a work around if that's the case.

Cheers!

santannamv, 2020-04-19:
Thanks by replying! I just updated to 2.82 and downloaded the last master and still getting the issue:

[image]

I hope it can help somehow in the add-on development. Also I expect to use it on windows! Thanks! 
Info about my system:

OS Name	Microsoft Windows 10 Enterprise
Version	10.0.17134 Build 17134
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	INSPIRON-15-500
System Manufacturer	Dell Inc.
System Model	Vostro 3478
System Type	x64-based PC
System SKU	0841
Processor	Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz, 1800 Mhz, 4 Core(s), 8 Logical Processor(s)
BIOS Version/Date	Dell Inc. 1.9.0, 04/07/2019
SMBIOS Version	2.8
Embedded Controller Version	255.255
BIOS Mode	Legacy
BaseBoard Manufacturer	Dell Inc.
BaseBoard Model	Not Available
BaseBoard Name	Base Board
Platform Role	Mobile
Secure Boot State	Unsupported
PCR7 Configuration	Binding Not Possible
Windows Directory	C:\WINDOWS
System Directory	C:\WINDOWS\system32
Boot Device	\Device\HarddiskVolume1
Locale	China
Hardware Abstraction Layer	Version = "10.0.17134.1098"
Username	INSPIRON-15-500\Administrator
Time Zone	Novosibirsk Standard Time
Installed Physical Memory (RAM)	8.00 GB
Total Physical Memory	7.77 GB
Available Physical Memory	2.77 GB
Total Virtual Memory	15.8 GB
Available Virtual Memory	8.26 GB
Page File Space	8.00 GB
Page File	E:\pagefile.sys
Kernel DMA Protection	Off
Virtualisation-based security	Not enabled
Device Encryption Support	Reasons for failed automatic device encryption: PCR7 binding is not supported, Hardware Security Test Interface failed and the device is not InstantGo
Hyper-V - VM Monitor Mode Extensions	Yes
Hyper-V - Second Level Address Translation Extensions	Yes
Hyper-V - Virtualisation Enabled in Firmware	Yes
Hyper-V - Data Execution Protection	Yes

Graphics:
Name	Radeon (TM) 520
PNP Device ID	PCI\VEN_1002&DEV_6665&SUBSYS_08411028&REV_C3\4&31323C1A&0&00E0
Adapter Type	AMD Radeon Graphics Processor (0x6665), Advanced Micro Devices, Inc. compatible
Adapter Description	Radeon (TM) 520
Adapter RAM	(2,147,483,648) bytes
Installed Drivers	C:\WINDOWS\System32\DriverStore\FileRepository\c0347105.inf_amd64_ae7d8883091aa10e\B347177\aticfx64.dll,C:\WINDOWS\System32\DriverStore\FileRepository\c0347105.inf_amd64_ae7d8883091aa10e\B347177\aticfx64.dll,C:\WINDOWS\System32\DriverStore\FileRepository\c0347105.inf_amd64_ae7d8883091aa10e\B347177\aticfx64.dll,C:\WINDOWS\System32\DriverStore\FileRepository\c0347105.inf_amd64_ae7d8883091aa10e\B347177\amdxc64.dll
Driver Version	26.20.13025.1004
INF File	oem69.inf (ati2mtag_R503DS section)
Colour Planes	Not Available
Colour Table Entries	Not Available
Resolution	Not Available
Bits/Pixel	Not Available
Memory Address	0xC0000000-0xCFFFFFFF
Memory Address	0xD0000000-0xD003FFFF
I/O Port	0x0000E000-0x0000E0FF
IRQ Channel	IRQ 4294967286
Driver	c:\windows\system32\driverstore\filerepository\c0347105.inf_amd64_ae7d8883091aa10e\b347177\atikmpag.sys (26.20.13025.1004, 584.41 KB (598,440 bytes), 30/09/2019 20:06)
	
Name	Intel(R) UHD Graphics 620
PNP Device ID	PCI\VEN_8086&DEV_5917&SUBSYS_08411028&REV_07\3&11583659&0&10
Adapter Type	Intel(R) UHD Graphics Family, Intel Corporation compatible
Adapter Description	Intel(R) UHD Graphics 620
Adapter RAM	1.00 GB (1,073,741,824 bytes)
Installed Drivers	C:\WINDOWS\System32\DriverStore\FileRepository\igdlh64.inf_amd64_061ec16507e674c7\igdumdim64.dll,C:\WINDOWS\System32\DriverStore\FileRepository\igdlh64.inf_amd64_061ec16507e674c7\igd10iumd64.dll,C:\WINDOWS\System32\DriverStore\FileRepository\igdlh64.inf_amd64_061ec16507e674c7\igd10iumd64.dll,C:\WINDOWS\System32\DriverStore\FileRepository\igdlh64.inf_amd64_061ec16507e674c7\igd12umd64.dll
Driver Version	22.20.16.4836
INF File	oem31.inf (iKBLD_w10_DS section)
Colour Planes	Not Available
Colour Table Entries	4294967296
Resolution	1366 x 768 x 60 hertz
Bits/Pixel	32
Memory Address	0xD1000000-0xD1FFFFFF
Memory Address	0xB0000000-0xBFFFFFFF
I/O Port	0x0000F000-0x0000F03F
IRQ Channel	IRQ 4294967288
I/O Port	0x000003B0-0x000003BB
I/O Port	0x000003C0-0x000003DF
Memory Address	0xA0000-0xBFFFF
Driver	c:\windows\system32\driverstore\filerepository\igdlh64.inf_amd64_061ec16507e674c7\igdkmd64.sys (22.20.16.4836, 12.80 MB (13,422,560 bytes), 06/11/2017 23:47)

kevancress, 2020-04-19:
Ah that's too bad. Okay I'll do some research this week and see if I can track down whats causing the shader compile error on your system and get it fixed! Thanks for all the info that really helps! Looks like it may be a difference between OpenGL support on NVIDIA and AMD cards since I've always been testing on a NVIDIA graphics card...

santannamv, 2020-04-20:
Take your time Kevan. I feel sorry I can not help. You guys are filling a huge gap. Congratulations!

kevancress, 2020-04-22:
Hey @santannamv 

I've just committed some changes to the shader that seems to be causing your issue. If you can give the newest master a try hopefully it will work!

If it still wont enable, can you try running Blender in the GPU debug mode, it should be a file called 'blender_debug_gpu.cmd' right below the main blender.exe wherever you have Blender installed.

Then try enabling MeasureIt-ARCH and then, if it fails, immediately close blender. It should give you a log in your temp files called 'blender_debug_output.txt' and if you can send me that file it will give me some more info about the details of the error that the graphics card is having with the shader!

Thanks!

[image]

santannamv, 2020-04-26:
Hi Kevan, sorry to respond that late. Here follow the log:

Switching to fully guarded memory allocator.
Blender 2.82 (sub 7)
Build: 2020-03-12 15:41:08 Windows Release
argv[0] = blender
argv[1] = --debug
argv[2] = --debug-gpu
argv[3] = --python-expr
argv[4] = import bpy; bpy.ops.wm.sysinfo(filepath=r'C:\Users\ADMINI~1\AppData\Local\Temp\blender\debug_logs\blender_system_info.txt')
Read prefs: C:\Users\Administrator\AppData\Roaming\Blender Foundation\Blender\2.82\config\userpref.blend
read file 
  Version 280 sub 39 date unknown hash unknown
GPUTexture: create : TEXTURE_CUBE_MAP, RGBA16F, w : 2, h : 2, d : 0, comp : 4, size : 0.00 MiB
GPUTexture: create : TEXTURE_1D, RGBA8, w : 1, h : 0, d : 0, comp : 4, size : 0.00 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1, h : 1, d : 0, comp : 4, size : 0.00 MiB
GPUTexture: create : TEXTURE_3D, RGBA8, w : 1, h : 1, d : 1, comp : 4, size : 0.00 MiB
Using OpenGL 4.3 debug facilities
GL application marker: Successfully hooked OpenGL debug callback.
found bundled python: C:\Program Files\Blender Foundation\Blender 2.82\2.82\python
GPUTexture: create : TEXTURE_2D, RGBA8, w : 961, h : 26, d : 0, comp : 4, size : 0.10 MiB
GPUTexture: create : TEXTURE_2D, R8, w : 16384, h : 128, d : 0, comp : 1, size : 2.00 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 405, h : 26, d : 0, comp : 4, size : 0.04 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1366, h : 20, d : 0, comp : 4, size : 0.10 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 237, h : 26, d : 0, comp : 4, size : 0.02 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 29, h : 446, d : 0, comp : 4, size : 0.05 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 208, h : 446, d : 0, comp : 4, size : 0.35 MiB
GPUTexture: create : TEXTURE_2D, R8, w : 16384, h : 128, d : 0, comp : 1, size : 2.00 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 237, h : 26, d : 0, comp : 4, size : 0.02 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 237, h : 198, d : 0, comp : 4, size : 0.18 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1128, h : 26, d : 0, comp : 4, size : 0.11 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1128, h : 27, d : 0, comp : 4, size : 0.12 MiB
GL API deprecated behavior: glLineWidth uses wide lines (parameter <width> with value '0x0') which are deprecated and do not work outside of the compatibility context
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1128, h : 26, d : 0, comp : 4, size : 0.11 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1128, h : 26, d : 0, comp : 4, size : 0.11 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 58, h : 591, d : 0, comp : 4, size : 0.13 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1128, h : 617, d : 0, comp : 4, size : 2.65 MiB
GPUTexture: create : TEXTURE_2D, DEPTH24_STENCIL8, w : 1128, h : 617, d : 0, comp : 1, size : 2.65 MiB
GPUTexture: create : TEXTURE_1D, RGBA8, w : 257, h : 0, d : 0, comp : 4, size : 0.00 MiB
GPUTexture: create : TEXTURE_1D, RGBA8, w : 256, h : 0, d : 0, comp : 4, size : 0.00 MiB
GPUTexture: create : TEXTURE_3D, RGBA8, w : 1, h : 1, d : 1, comp : 4, size : 0.00 MiB
GPUTexture: create : TEXTURE_1D, RGBA8, w : 1, h : 0, d : 0, comp : 4, size : 0.00 MiB
GPUTexture: create : TEXTURE_2D, R11F_G11F_B10F, w : 1128, h : 617, d : 0, comp : 3, size : 2.65 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1128, h : 617, d : 0, comp : 4, size : 2.65 MiB
GPUTexture: create : TEXTURE_2D, R32UI, w : 1128, h : 617, d : 0, comp : 1, size : 2.65 MiB
GPUTexture: create : TEXTURE_2D, RG16, w : 1128, h : 617, d : 0, comp : 2, size : 2.65 MiB
GPUTexture: create : TEXTURE_2D, RGBA16F, w : 64, h : 64, d : 0, comp : 4, size : 0.06 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1128, h : 617, d : 0, comp : 4, size : 2.65 MiB
GPUTexture: create : TEXTURE_2D, DEPTH24_STENCIL8, w : 1128, h : 617, d : 0, comp : 1, size : 2.65 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1, h : 1, d : 0, comp : 4, size : 0.00 MiB
GPUTexture: create : TEXTURE_2D, DEPTH_COMPONENT24, w : 1, h : 1, d : 0, comp : 1, size : 0.00 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1128, h : 617, d : 0, comp : 4, size : 2.65 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1128, h : 617, d : 0, comp : 4, size : 2.65 MiB
GPUTexture: create : TEXTURE_2D, DEPTH24_STENCIL8, w : 1128, h : 617, d : 0, comp : 1, size : 2.65 MiB
GPUTexture: create : TEXTURE_2D, R16UI, w : 1128, h : 617, d : 0, comp : 1, size : 1.33 MiB
GPUTexture: create : TEXTURE_2D, DEPTH24_STENCIL8, w : 1128, h : 617, d : 0, comp : 1, size : 2.65 MiB
GPUTexture: create : TEXTURE_2D, R8, w : 16384, h : 64, d : 0, comp : 1, size : 1.00 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 527, h : 491, d : 0, comp : 4, size : 0.99 MiB
read file 
  Version 280 sub 39 date unknown hash unknown
GPUTexture: create : TEXTURE_2D, RGBA8, w : 961, h : 26, d : 0, comp : 4, size : 0.10 MiB
GPUTexture: create : TEXTURE_2D, R8, w : 16384, h : 128, d : 0, comp : 1, size : 2.00 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 405, h : 26, d : 0, comp : 4, size : 0.04 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 1366, h : 20, d : 0, comp : 4, size : 0.10 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 237, h : 26, d : 0, comp : 4, size : 0.02 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 29, h : 446, d : 0, comp : 4, size : 0.05 MiB
GPUTexture: create : TEXTURE_2D, RGBA8, w : 208, h : 446, d : 0, comp : 4, size : 0.35 MiB
GPUTexture: create : TEXTURE_2D, R8, w : 16384, h : 128, d : 0, comp : 1, size : 2.00 MiB
[... cut ...]
    from .measureit_arch_geometry import clear_batches, draw_annotation, draw_arcDimension, draw_alignedDimension, draw_line_group, draw_angleDimension, update_text, draw_axisDimension, draw_boundsDimension, get_mesh_vertices, printTime
  File "C:\Users\Administrator\AppData\Roaming\Blender Foundation\Blender\2.82\scripts\addons\MeasureIt-ARCH-master\measureit_arch_geometry.py", line 65, in <module>
    geocode=Line_Group_Shader_3D.geometry_shader)
Exception: Shader Compile Error, see console for more details

Blender quit
Press any key to exit . . .

kevancress, 2020-04-26:
Okay! I think I've fixed the errors that are showing up in the log. I've just committed them to the master branch so if you could download it and give it a try it should be fixed!

If it still fails to enable could you run blender in the gpu debug mode and post the log again.
 
Thanks for testing this out for me, and posting the logs! I really appreciate it since I've got no way to test on other graphics cards here!

santannamv, 2020-04-27:
Thanks a lot Kevan! Seems to work smoothly now. In few months I will adopt blender and add-ons like yours in a lab with 50 pc`s. Eventually I will report about any issues. Have a nice week!

kevancress, 2020-04-27:
Wow that's great to hear! Happy to help whenever there's an issue! Let me know too if there's any features or tweaks to MeasureIt-ARCH that would help with the workflow in your lab! Cheers!

closed 2020-04-27 by kevancress