kevancress/MeasureIt_ARCH #58: Can't render image

meutzitzu, 2020-04-20:
I have this big Blender file with multiple scenes I'm working on and I've used MeasureIt-ARCH for a while and it's been working fine, but now it won't render a measureit and it just throws out this error

Traceback (most recent call last):
  File "/home/meutzitzu/.config/blender/2.82/scripts/addons/MeasureIt-ARCH-master/measureit_arch_render.py", line 121, in execute
    if render_main(self, context) is True:
  File "/home/meutzitzu/.config/blender/2.82/scripts/addons/MeasureIt-ARCH-master/measureit_arch_render.py", line 298, in render_main
    draw_axisDimension(context, myobj, measureGen,dim,mat)
  File "/home/meutzitzu/.config/blender/2.82/scripts/addons/MeasureIt-ARCH-master/measureit_arch_geometry.py", line 451, in draw_axisDimension
    p1 = get_point(obvertA[dim.dimPointA], dim.dimObjectA,aMatrix)
IndexError: list index out of range

location: <unknown location>:-1

(running on Kubuntu 19.10 btw )

MeasureIt-ARCH works just fine for other files but I can NOT lose all the stuff I made in that file 
I'm hoping there's a fix for this since appending all of that stuff into a new file would be pretty inconvenient

kevancress, 2020-04-21:
Hey @meutzitzu 

Sometimes this shows up if there's a corrupted dimension in the scene who's anchor vertex no longer exists. Usually MeasureIt-ARCH cleans these up automatically when you save and load a file but a few seem to slip through my checks. There's a couple options to fix it.

1) If everything is still displaying how you want it too in the viewport view, then I think if I just implement some proper error handling in the code then MeasureIt-ARCH should catch the error and just keep on going. That should let the render continue normally.

If it's not displaying properly in the viewport either then we'll have to fix the dimension manually I think.

In that case if you can send over the file I can probably find the offending dimension pretty quick using the VS code debug tools and fix it up for ya! But I'll try out the error handling option first today and see if that sorts things out! I'll post again here once I've got that committed to the master!

PS. I've been messing around with the SolveSpace tool you posted the other day and It's great!! Love the 'drag to move' dimension text, definitely hoping to add something like that to MeasureIt-ARCH with the gizmo system soon, and maybe a toggle in the styles for screen space text?

meutzitzu, 2020-04-21:
https://drive.google.com/open?id=1_YexTxG0SBPHUJ7LCjEMNq6RSy85vpKt

okay so here's the file 
just fair warning : there are a metric crapton of dimensions within 3 scenes ... it's kinda huge

thanks in advance. 

also, yeah, upon finding SolveSpace I started to switch away from FreeCAD it's SO GOOD 

i mean don't get me wrong, FreeCAD is awesome, it's the most complete open source CAD program there is. but sometimes you just want that quick adaptor you want to 3d print or CNC in a hurry and for that SolveSpace is a godsend

kevancress, 2020-04-22:
Okay! So I think I've got this one sorted out! Thanks for sending the file over that helped a ton!

I added some new error handling that should automatically clean up and delete any of the corrupted dimensions and annotations that were stopping the render and slowing down the scene!

So if you download and install the latest master all you should have to do is open up your file, show all of the dimensions then just pan around the scene a little and it should clean them all up and render without any problems (at least it did when I opened the scene up here)! 

Also I noticed you'd been using an older version of the add-on, so the new master version will probably break your styles ... sorry ... but if you make a new style with the same name as the old one the dimensions should automatically link back up to it! (I had to break some backwards compatibility to get a pretty significant performance improvement, so hopefully the scene should be faster at least)

There's a couple other new features in the new version (annotations support more than one line of text now, text should automatically orient itself to be readable, and probably a couple other things) Hopefully it doesn't break too much!

Let me know how it works in the new version! If anything seems too different/ breaks workflow between the old version and the new master let me know and I'll see what I can do!

meutzitzu, 2020-04-24:
stupid question, but 
how do I install the new master ? 

I used to use the zip files under the release section in the readme

kevancress, 2020-04-24:
Oh right sorry I should have explained! On the main page  click on the clone or download button in the right corner and then click download zip, that'll always downloaded the latest master version of MeasureIt-ARCH even if I haven't made an official release. Then you can install it like you would the regular release

[image]

You'll probably have to remove the old version and restart blender before installing the new version too, if it's not removed and restarted first then sometimes it can cause problems!

[image]

meutzitzu, 2020-04-25:
YUP, it worked, thank you SO MUCH

kevancress, 2020-04-25:
Awesome!! Glad I could help!

closed 2020-04-25 by kevancress