kevancress/MeasureIt_ARCH #60: collection bound dimension bug

Elwex123, 2020-05-03:
I added a UV sphere to a collection bound dimension, and when I rotated it the dimension updated the wrong way. I attached a clip showing the problem.

[image]

Elwex123, 2020-05-03:
When I apply rotation it fixes itself.

kevancress, 2020-05-03:
Hey @Elwex123,

Yeah I've been trying to find a solve for this one for a bit now. It's happening because I'm using Blender's built in bounding boxes and finding the extremes of them to work out the bounds measurements, but blender's bounding boxes use the local transform of the object and don't stay aligned to the world x,y, and z axis (gif below), that's why it corrects itself once the rotation is applied. I could actually loop through all the geometry to find the bounding box, but this gets pretty slow in python if there's a lot of geometry... maybe I could find a way to undo the rotation of the object's local transform when checking the bounding box points (sort of 'applying the rotation' but only temporarily for the bounding box) ... anyway I'll take a look at it this week and see if I can come up with a solution.

Cheers!

[image]

kevancress, 2020-05-03:
@Elwex123  Okay I think I've got a fix working that shouldn't slow things down too much, it's in the latest master if you want to give it a test!

Essentially it check each object in the collection for any local rotation. If it has rotation, it does a full calculation to find the world space bounding box for that object, otherwise it just uses blenders default bounding box, so hopefully it should perform okay.

I also made a change so that bounds dimensions will stick to the objects local space bounding box by default, but added an option to always calculate the world space (axis aligned) bounding box if that's preferable. (gif below)

[image]

Elwex123, 2020-05-08:
Well, I downloaded the latest master today and didn't find the option to always use axis aligned bound. Maybe I'm missing something?

Elwex123, 2020-05-08:
Oh, found it! Apparently blender needs few restarts or something to update, IDK...
Well, I found 2 bugs:
A. Measureit Arch does not assign styles to bounding boxes when they are created. 
B. Collection bounding box freaks out when object is scaled by negative numbers.
[image]

Elwex123, 2020-05-10:
I also noticed that bound dimensions don't align to view planes.
[image]

kevancress, 2020-05-11:
Hmm... Okay, I'll take a look into the negative scaling issue this week, not quite sure what could be causing it...

The styles issue I should be able to fix up today. 

The view plane placement might take a bit longer, I've been trying to work out what the best automatic placement system would be for bounds dimensions. I'm thinking that rather than shifting view planes the dimensions themselves should move so that the bounds dimensions always appear on the left and top edges of the bounding box relative to the current view?

kevancress, 2020-05-15:
Okay! so the style assignment and scale issues should be fixed in the latest master!

I'm gonna close this issue and move the view plane placement issue to a new thread to keep things organized, hope thats okay!

closed 2020-05-15 by kevancress