kevancress/MeasureIt_ARCH #61: Dimensions between vertices of two different objects

c1symeon, 2020-05-11:
I think this would be useful.  In your example of the house the wall needs to be one object for the dimension workflow shown (including cutouts for windows that cannot be boolean subtraction modifiers).  In practice this is a restricting condition.
For more freedom in our definitions of objects (be it for hierarchy, materials, different modifiers, etc), I think dimensions must be drawn between selected vertices of different objects. 
Also consider freeing up the dimension reference points from vertices.  This would be useful if for example you wanted a clearance of 30cm from an object, you can dimension it, the dimension will belong to the object, but the one end of the dimension would be set independently

kevancress, 2020-05-11:
Hey @c1symeon, thanks for the feedback! MeasureIt-ARCH actually does support dimensions between  vertices of two different objects (although the workflow isn't really clear, I should update the docs and examples to make it easier to find. Example gif below). I think I need to work on an operator to make this process simpler too, since going into edit mode on each object separately is a bit of a pain.

For the case of dimensions that are selectively attached to vertices, you could add a MeasureIt-ARCH dimension to two empty objects and use either parenting, vertex parenting or Blender's hook constraints to tie one empty to the object, while the other could be placed freely!

[image]
 
Hope that helps!

c1symeon, 2020-05-13:
Yea, it does thank you.  I found out in the meantime how it works.
Btw, you don't need to go into Edit Mode separately, just select both and go into edit mode (Blender 2.8x allows you to edit many objects simultaneously and it works with your tool) and select one vertex on each.  Then tab out and ask for the dimension.
I don't know why you have limited to one vertex for each though, I suggest you use the geometric mean of all vertices, when more than one is selected, that would work very well if you wanted to draw a dimension from a point of one object to the middle of an edge (two points selected) or to the center of mass (8 points of a cube selected).  Whether the selection of many points has meaning or not is up to the user.  Surely you have used the above two selections techniques when modeling, do the same for dimensioning.

kevancress, 2020-05-13:
Thats a great suggestion!! I'll work on adding that in for the next release!