kevancress/MeasureIt_ARCH #65: Add Multi Line Text & Custom Text for Dimension Elements

kevancress, 2020-05-15:


meutzitzu, 2020-05-31:
Does this mean we can use Unicode like in FreeCAD where whenever I want to add a diameter dimension I just use the good'ol Ctrl+Shift+U, 2, 2, 0, 5  ( ∅ )

kevancress, 2021-03-04:
Finally got a start on adding this in dc3a323b46ae3ad4d604ab4deb908bfc4c315728. Just a Single line off custom text that overrides  the default measurement as a start but I'll work on adding multi line, and prefix/suffix custom text too!

kevancress, 2021-05-19:
Closing this, custom and multiline dim text is supported now!

@meutzitzu for some reason it won't let me type alt codes for special characters directly in blenders text boxes, but the do work if you copy and paste them in

[image]

closed 2021-05-19 by kevancress