kevancress/MeasureIt_ARCH #66: Dimension element freaks out when shrinkwarped

Elwex123, 2020-05-20:
I made an edge with dimension element and shrinkwraped it to an object. In the previous version (3) it worked just fine, but for some reason in this version (.4 dev) it freaks out: 
[image]

Also, for some reason Measureit-ARCH doesn't recognize the 2 verts I'm selecting if one of them isn't active (highlighted in white). From what I can recall, I think it worked in version 3, but I may be wrong on that one.

kevancress, 2020-05-21:
Hey @Elwex123, good catch, thanks!

Should be fixed in the latest master, I had been trying out a different method for evaluating modifiers, and it looks like it wasn't as stable as the old approach, so I've switched back.

Your right about the selection issue, I've been making some changes to how selections are interpreted for dimension creation, the upside is that the dimensions created more accurately reflect the order that the user selects the vertices in (gif below), but the downside is that box select and circle select no longer work in vertex selection mode, because they don't keep track of the selection order, I've updated the error message to be a little more clear. (both circle and box select still fine in edge select mode though)

[image]
.

Elwex123, 2020-05-21:
Yep, shrinkwarpping works like a charm! And the multi edge selection is really handy, thanks for the quick fix!

closed 2020-05-22 by kevancress