kevancress/MeasureIt_ARCH #67: Dimension elements bug when multi-select applyed to duplicated edges

Elwex123, 2020-05-21:
This is a weird bug, I would say... You take a mesh, duplicate an edge, separate it to be a different object, in edit mode duplicate the edge and apply dimension elements to both at the same time (with edge select mode). After that, this object becomes buggy and unusable. Dimension elements don't appear on the dimension elements list, and it always shows error when I try to add new dimension measurements to this specific corrupted object. 
I use 2.82, if that's important.
[image]

kevancress, 2020-05-22:
Ahh looks like I wasn't refreshing the edit mode mesh data before creating dimensions, so it was giving an error when trying to dimension any geometry that had been created after you entered edit mode!

Just pushed a fix to the latest master show it should be fixed now!

Also, in case an object gets corrupted like this due to some other error in the future, the 'delete all  dimensions' operator in the drop down menu next to the list of dimensions should clear out any buggy dimensions, even if they aren't showing up in the list.

Elwex123, 2020-05-24:
Looks like problem solved. Thanks for the quick fix!
I'll let you know if I'll encounter more bugs like this.

closed 2020-05-24 by kevancress