kevancress/MeasureIt_ARCH #68: Font Size unit for dimensions (and maybe all annotations)

c1symeon, 2020-05-31:
What does the font size mean?
In MeasureIt the font size is in pixels.  As a result when you zoom in and out of the screen, the font stays the same size and becomes smaller and larger compared to the objects respectively (always stays at pixel size on screen).
This can be used when you print annotations also, for example: setting font size at 24pixels at 300 dpi, means that when you print 100% (1:1) your font will be about 2mm (300 dpi = 12 dots per mm, means 2mm height font for 24pixels).
Unlike this predictable behavior, your Add-on creates text that scales with the objects when zooming in the viewport.  What does a font size like 24 mean in this case? How do I control font size in the render image?

meutzitzu, 2020-05-31:
I believe it uses Scene Units as the suffix.
*Scene > Units > separate units* and then *Length: *
note when *Rotation* set to radians it automatically truncates it to the same precision as the other values. For my use case, since I normally use a precision of 0 and use centimeters for everything, the radian angle gets hillariously truncated to the first integer so it can only take the values 1, 2 and 3 XD. well obvious things can be done about this, one would be to add a math evaluator that reduces the value to fractions of pi, or to just add a different precision slider for radians, or (and this is the best one) just to not bother, at least for now because who uses radians in blueprints lol ?

c1symeon, 2020-06-01:
@meutzitzu : You are missing my main issue, your reply partially answers the last sentence only.  It is clear that the units are the scene units, the question there was one of comparison with Measureit, where you can have the unit as a suffix, you can specify another unit (and the dimensioning addon does the conversion math, without the altering the scene units) or you can just have the number with no suffix, which actually is the most common case, one rarely sees a technical drawing with dozens dimensions repeating "m", for meters in every one.
The main issue however was about the actual size of the letters of the dimension annotation (not the dimension distance)

kevancress, 2020-06-02:
Hey all,

@c1symeon  your right the font size value is totally arbitrary right now because I'm placing objects in world space rather than screen space like the original measureIt did. 

To have a meaningful font size value I need some real world print space or paper space defined to base that font size value on. Eventually I'm hoping to merge the per_camera_resolution addon with MeasureIt-ARCH that way I can use the dpi and drawing scale factors set for the active camera to allow users to define font size with real units (pt's, mm's or in) but that may take some time.

Also I'm going to create a new issue for adding an option to display dimensions with out the suffix, that should be a quick one to implement!

@meutzitzu also going to create a separate issue for an option to set radians precision separately  as fractions of pi, (probably a pretty niche use case your right, but its a quick fix)

kevancress, 2020-07-26:
Hey all,
Quick update on this since recent development has made it possible to actually make some improvements here!

Per_Camera_Resolution was merged with MeasureIt-ARCH a few weeks back allowing for paper outputs to be defined in real world units, and i've just finished a refactor of the text system to allow for meaningful font values.

as of c17c68c2e3087cfdd01b71014055dca734918fb3 font size is now defined in pt's (1pt = 1/72") and is determined based on the current view's scale (or the default scene scale if no views have been created). This is applied for every MeasureIt-ARCH element that uses text, so all dimensions and annotations have real font units now.

[image]

kevancress, 2020-07-26:
Closing for now since Units have been implemented.

Feel free to re-open or start a new issue if you have any comments or suggestions on how the current implementation functions

closed 2020-07-26 by kevancress