kevancress/MeasureIt_ARCH #70: Unstable: Loss of dimensions and GUI messup

c1symeon, 2020-06-01:
Hi,
I tried to use more rigorously to add dimensions to a small plan, a bathroom layout.
Seemed to work ok.  After one save the Dimension style was lost (the link references in the object properties turned red).  Redefined the style and briefly the dimensions seemed ok.
After the next save I lost dimensions of several objects, retained only one.
Trying to manipulate the properties, messed my GUI: the letters in Blender menus turned fuzzy and changing between workspaces did not regenerate all the windows, but rather overlayed some over others randomly.

kevancress, 2020-06-02:
Hey @c1symeon, thanks for the report a couple of questions to help me track this down:

1) What version of blender were you using at the time?
2) What Operating system and graphics card are you using?
3) Do any error messages appear in blenders console (window -> toggle system console) when opening a file where styles of dimensions are lost, or when the issues occur with the gui?
4) Did you change version of MeasureIt-ARCH between saving and loading the files (the code for the style system has changed from v0.3 to the current master, so styles in files saved with v0.3 will be broken when opened with the current master version.

re: corrupted dimensions - MeasureIt-ARCH dimensions become corrupted if the vertex index that they reference no longer exists, this can happen through changes to the mesh in edit mode or by applying some generative/destructive modifiers, in these cases, MeasureIt-ARCH automatically deletes the dimension if it can't find the vertex index it should be attached to. This likely isn't the case in this instance as no changes to the mesh would have occurred between saving and loading the file.

If your able to create and post a file where this is reproducible, or post the error messages that appear in the blender console that helps a lot!

kevancress, 2020-07-26:
Closing this for now since I haven't been able to reproduce the issue and need more info to be able to look into it.

@c1symeon feel free to re-open with the above info if you're still experiencing this problem.

closed 2020-07-26 by kevancress