kevancress/MeasureIt_ARCH #75: v0.3: Annotation camera view association missing

globalcitizen, 2020-06-02:
There is no camera view association feature for annotations, only for dimensions.

Since for real world use (generation of plans and elevations from a model) the association of these labels is drawing (=camera view) specific, this is a serious missing feature.

kevancress, 2020-06-03:
@globalcitizen This would actually be a great first issue to get familiar with MeasureIt-ARCH's code if you want. The property that controls the camera association is the 'dimVisibleInView' property that you can see in 'measureit_arch_baseclass.py' right now only dimensions inherit that property from the BaseDim class, but if you move it up to the BaseWithText class to let annotations inherit it as well.

From there you can see how its used in the dimension UI (draw_aligned_dimension_settings in 'measureit_arch_dimensions.py') and in the draw code ('draw_alignedDimension' in measureit_arch_geometry.py) to control visibility and port that over to the annotation UI and draw code.

Actually it might be worth adding it to the line groups as well since you might want to change line styles between views?

globalcitizen, 2020-06-03:
Thanks for describing this first issue. Moving the property to the superclass sounds a reasonable and straightforward fix.

Before diving in to code I would like to do a bit of survey work on what's the current state before jumping in. Today I will focus today on evaluating BlenderBIM's implementation as-is, which may already be adequate for my purposes. While doing that I will try to write up some notes.

This seems the right way to proceed for two reasons:
- I am not really experienced with Python (although I have written production code in a great many interpreted languages) getting going will likely take a couple of days
 - Focusing purely on my immediate motivations, I actually have an immediate need for drawings within the next couple of weeks, which may be met by BlenderBIM's current code with some pretty high hassle workarounds.

Based on the notes I come up with and whatever you and @moult think, perhaps we can hatch a plan to combine the best features of all existing codebases (possibly including your other modules you mentioned you may wish to merge) in to a combined project, which in getting a few developers on one codebase could make things significantly easier. I would be very happy to contribute to this as I think the work you have both done is awesome and deserving of greater attention.

kevancress, 2020-06-03:
That sounds like a great workplan to me! 

Having your notes on both add-ons would be very useful going forward(and the feedback you've already given has been fantastic!)

In that case I'll wrap up this feature and create a new dedicated thread where we can keep notes and have a more general discussion about the potential collaboration & combination of MeasureIt-ARCH and Blender BIM

kevancress, 2020-06-03:
Visible in view added to all MeasureIt-ARCH element types in 0e8d237a6b5237448864c57144dee8466e4642a6

closed 2020-06-03 by kevancress