kevancress/MeasureIt_ARCH #76: v0.3: Example output process undocumented

globalcitizen, 2020-06-02:
It would be far more useful for random users and probably help them to understand the how and the why of the addon to give some video and image examples of how to actually use the output from the addon in a real world context, ie. to generate dimensioned plans and elevations.

I would suggest therefore when making the 0.4 release video:
- Take a monkey or cube
- Dimension it using the new v0.4 major dimension simultaneous multi-dimensioning tool
- Output this (via SVG?) to a useful place (ie. actually make some plans in inkscape with standard architectural visual format / layout including job, drawing name, drawing number, north point, etc. and transfer the output there) preferably using a specific scale (eg. 1:100)
- Save as PDF

Personally I am not quite sure how to do that preserving data in vector (eg. via SVG export) though it should be possible I suppose?

globalcitizen, 2020-06-02:
Update: I saw your comment [over here](https://www.youtube.com/watch?v=lHI78SDB8bs) regarding your other [per-camera-resolution addon](https://github.com/kevancress/per_camera_resolution). An example workflow should be documented as this apparently is part of the recommended process.

kevancress, 2020-06-02:
Yep a more real-world applicable demo video is definitely in the works, and I'm hoping to actually merge the per-camera-resolution addon with MeasureIt-ARCH eventually to make workflow better.

Right now I'm limited to raster outputs, SVG is definitely a long term goal, but I need to familiarize myself with the standard & learn how to write an exporter first. The other roadblock there is that currently a lot of MeasureIt-ARCH's visuals are computed in shader on the GPU for performance reasons, so code for an svg exporter would need to be kept in sync with the shaders to ensure a reasonably similar output.

Ideally I'd like to have an system for title blocks,drawing layout, scheduling, systems directly in blender, tied to the camera ( a sort of CAD style paper space almost) but again that's a fairly long project.

globalcitizen, 2020-06-03:
On a related note (re: vector output achievement) I had a chat with @Moult about [BlenderBIM](https://blenderbim.org/) and for that project he has only used vector output in some situations also. Basically, even if you use BlenderBIM to export in SVG, within the SVG is going to be an embedded bitmap image... possibly some of the time with varying amounts of vector elements drawn on top. The whole export kafuzzle runs on some kind of external C++ based IFC library and exports are slow with large volumes of data which were factors leading to the current hybrid output strategy. On the plus side, they do have sections and cameras already, which is cool. It would be great if both projects could combine efforts to prevent duplication of effort. I will try to contribute too, after taking a deeper look tomorrow (was offline today).

kevancress, 2020-06-03:
Yeah I was just taking a look at the latest version of BlenderBIM last night and it looks fantastic! I definitely think a collaboration could be a huge step forward. @Moult has done such great work giving blender the capability to support and interface with these industry standard formats and connecting measureit-arch with that would be amazing! 

It would be great if you want contribute too! I'll try and go through and mark some of the issue reports that I could use a hand with today!

kevancress, 2020-06-06:
Alright so some proof of concept work on this.

I've merged per camera resolution with MeasureIt-ARCH and refactored it to be a list based UI for selecting and defining views. I've also started started to port some of BlenderBIMs svg export code over to work with MeasureIt-ARCH's drawing system. Its just basic linework for now (text and polygons to come) and I'll need to sort out some way to do vector depth sorting and occlusion for hidden linework and all that... but its a start!

[image]

globalcitizen, 2020-06-06:
__Awesome!__

So ... are you planning to port the hatch support to support industry standard patterns for concrete, glass, etc? Then there's the missing text, sectioning support, options for bitmap-in-SVG, and drawing scale. Once those are done, this module will be _much_ easier to point people to than the BIM one, and easier to learn.

Regarding the vector depth sorting and occlusion, shouldn't it be possible to sort visible objects by camera-to-object depth, then render them in reverse order (towards the camera) to automatically provide correct Z-axis ordering of resulting vector shapes?

Moult, 2020-07-27:
[image]

I'm struggling to maintain consistent font sizes. I'm not sure what affects font size. In CAD detailing, font sizes are usually not configured on a case by case basis. Usually, there are only 5 standardised technical lettering sizes. For more professional output, perhaps it is a good idea to disallow users to configure on a case by case basis by default, and come with these presets preloaded?

Feel free to steal my font size code and font - happy to walk you through how it works :)

JanFilipec, 2021-11-29:
Hi, it seems that no matter what I do, the units in the exported svg are inches:

[image]

which then leads to imprecise results while using metric (I set the paper size to 1 by 1 meter in Blender):

[image]

am I missing something?