kevancress/MeasureIt_ARCH #77: v0.3: Default 45 degree offset for vertices is inconvenient

globalcitizen, 2020-06-02:
The default of a 45 degree offset for a dimension of two vertices on a single plane is inconvenient for both plans and elevations, which by far represent the most popular output types.

[image]

Unsure of this is fixed in the planned v0.4 release as I am unable to get the direct from source version working (tried zipping the Github repo, for some reason Blender wouldn't recognize it).

kevancress, 2020-06-02:
@globalcitizen  Yep this default behavior has been changed in v0.4 + to automatically align with a plan or elevation view depending on the users viewpoint

hmm... strange that the zipped repo isn't being recognized... I just put up a release for version 0.4 so hopefully that installs correctly, let me know!

If your interested in keeping up to date with the latest master and reloading the add-on directly from the cloned repo I've been using Jacques Lucke's visual studio code extension for Blender addon development, it's worth checking out: https://marketplace.visualstudio.com/items?itemName=JacquesLucke.blender-development

globalcitizen, 2020-06-03:
Tried to install v0.4 beta source code zip file `MeasureIt-ARCH-mARCHv0.4.zip` and got this error.

[image]

closed 2020-06-03 by globalcitizen