kevancress/MeasureIt_ARCH #78: v0.3: Auto-hide, alternatively display or realign depth axis labels

globalcitizen, 2020-06-02:
The following is a top orthographic view of a default cube. The dimension for the Z (depth) axis shows as a 45 degree line.

[image]

While the issues #74 and #75 would progress toward fixing this, a quicker option saving user time would be to treat this as a detectable condition and either auto-hide it or create some form of alternate display.

I imagined there is a depth indicator symbol in engineering symbology already and discovered ↧ which is apparently adequate and used similarly to the diameter symbol. Changing the dimension to an annotation and adding ↧ as a prefix would be one option for automatically resolving these issues. The outstanding question is then how to perform programmatic detection.

kevancress, 2020-06-02:
I think #74 and #75 are definitely the quickest fixes here. 

Detecting this as a condition and automatically generating an annotation would be fairly complex to implement but is an interesting possibility for future development!

(Also I'm  removing the v0.3 from the title since this still effects the latest version)

meutzitzu, 2020-06-02:
one way to display depth is to use a physics style normal vector notation ( a circle and a point meaning from the plane towards the camera and a circle with an X meaning away from the camera ) 
I don't think this is actually used in engineering/architecture though

globalcitizen, 2020-06-03:
Agreed the cultural convention is important and the original suggestion of ↧ is no good in this regard. From what I can gather online, for engineering professions the use of ↧ is legible to people boring holes from shop plans. However, this may not extend to all construction workers and non hole-boring-related depth specifications.