kevancress/MeasureIt_ARCH #82: v0.4 beta: Zipfile installs but addon does not activate

globalcitizen, 2020-06-03:
Tried to install v0.4 beta source code zip file `MeasureIt-ARCH-mARCHv0.4.zip`. Install worked fine, the module appeared in the available modules list. However, when I enable it by clicking the checkbox I get this error:

[image]

kevancress, 2020-06-03:
ah sorry that's a naming error, the folder inside the .zip shouldn't have had a "." in the name. I've updated the zip file on the release page.

If you remove the version that you have installed, restart blender, and then install the new .zip package from the [release page](https://github.com/kevancress/MeasureIt-ARCH/releases/tag/mARCHv0.4) called "MeasureIt-ARCH_v04.zip" it should work fine.

globalcitizen, 2020-06-03:
Resolved.

closed 2020-06-03 by globalcitizen