kevancress/MeasureIt_ARCH #83: v0.4: Text illegibly small by default

globalcitizen, 2020-06-03:
After adding major dimensions for an object using the new Bounds feature, they were illegible even after zooming in.

[image]

globalcitizen, 2020-06-03:
Probably related to #68 but I feel this actually went backward since v0.3. Maybe due to ongoing work related to #68?

kevancress, 2020-06-03:
Yep, definitely related to #68, having a sensible approach to font size without a defined output (paper size, drawing scale etc.) has led to text being a bit arbitrary right now.

for axis and aligned dimensions the initial font size is actually proportional to the length of the dimension on creation, which leads to reasonably use-able results in the viewport, but really its a bit nonsense and impractical for a professional workflow.

for now I've updated the master branch 73d90f9c85a2395f49a463a08e99b360c6db4383 with some more sensible default values for the initial creation of bounds text (centered, font size 55)... perhaps I'll take a stab at finally merging the per camera resolution module in to this over the next day or two, since not having a 'paper space' defined is starting to be quite a hindrance for issues like this

kevancress, 2020-07-26:
Closing this since it should be resolved now that #68 has been fixed.

closed 2020-07-26 by kevancress