kevancress/MeasureIt_ARCH #84: v0.4: Vertices selection apparently buggy

globalcitizen, 2020-06-03:
After selecting two vertices in edit mode and pressing the Axis button, I received an error as follows.

[image]

kevancress, 2020-06-03:
Ahh right I need to update the tooltip for this to make it more clear.

The gist of the issue is that MeasureIt-ARCH uses the order the vertices are selected in when creating dimensions in vertex select mode. Since the circle and box select modes in blender don't keep track of selection order, they cause an error when used to select verts for measureit-arch dimensions. (there's a little more detail [here](https://github.com/kevancress/MeasureIt-ARCH/issues/66#issuecomment-631877165))

globalcitizen, 2020-06-03:
Haha, that makes perfect sense as a programmer but as a user it is not just unusable and misleading it is silently broken since the last version.

I would suggest the following fix for the UX:
- Accept the unordered vertices as input
- Try to automatically determine a reasonable default orientation (order of vertices), for example based upon (1) which way is up (2) which way the nearest dimensions are oriented (3) which way most dimensions are oriented.
- Give a quick reshuffle button to toggle the display to an alternate vertices order if the user realises the assumed order is incorrect. With a max of two more button pushes, this seems a very quick and complete workflow, and will work out faster than ordered selection with less of a learning curve for the user and less mental overhead.