kevancress/MeasureIt_ARCH #85: v4.0: Dimension Font Crop,Theme goes crazy,Line Anti-aliasing,Viewport vs Render difference

tilakji, 2020-06-05:
First of all,Kudos to Antonio Vazquez, kevancress (you) and all of the blender developers and add-ons developers for it.
Thanks!

B3D v2.81.16 | add-on v4.0
issues:
• Dimension Font gets cropped.  (ref font: flux architect)
[image]

• Theme goes crazy after hitting "Measure-it ARCH Image" when used in a project that had around 50k faces
-Before
[image]
-After
[image]
(UI Text/theme render-flaw happens)

improvement:
• Need better anti-aliasing
[image]

• Dimension position differs, comparing Viewport vs Dimension Render. 
-Viewport:
[image]
-Dimension Render Composite:
[image]
Please clarify if that is intended to work like that. or we have option to tweak that?

Thanks!

kevancress, 2020-06-05:
Hey @tilakji Thanks for the feedback!

I've just pushed a fix first two major issues, (text clipping and UI error) and the imperial rounding error so they should be resolved in the latest master. If you can give it a test and let me know if the fix works for you that would be great!

For the other issues:

Anti-aliasing: A local AA solution for measureIt-ARCH is in the works, you can see how its coming along if you look at the AA on the line group lines. Right now my implementation breaks down for thin lines, so I haven't implemented it for dimension leader lines yet. I'm taking a lot of inspiration from the [shapes library](https://twitter.com/FreyaHolmer/status/1255193781739180032) being developed for unity right now, so hopefully we'll get to that level of line quality soon.

Render vs. Viewport position: Right now, if you don't have a view plane specified, dimensions are positioned based on the users viewpoint in the 3d scene. So a general perspective view won't necessarily match the render and that's intentional to try and keep the dimension legible when working in the viewport. That said, the dimensions position in the camera preview view should always match the render so you can accurately preview the drawing you want to export. If it doesn't match the camera view let me know and I'll try and get that fixed.

kevancress, 2020-06-09:
Hey @tilakji, I'm closing this issue for now since I think the Major Problems you identified are resolved. 

re Render vs. Viewport position: If this is still an issue based on the comments above, please let me know and open a new issue report for it, Thanks!

closed 2020-06-09 by kevancress

tilakji, 2020-06-12:
hello @kevancress ! sorry for late reply,stuck somewhere.
just downloaded 2.83 LTS, i will check our MeasureIt-Arch's latest release with that and report to you.
Thanks!!!

HarryC1, 2020-06-13:
Hi
0.4 shows image shows a dimension but I still can’t get I right. I’m waiting for the video which might describe how to get the dimension and the arch.