kevancress/MeasureIt_ARCH #86: Angle and arc not working.

HarryC1, 2020-06-06:
Angle and arc not working.
Performed as per manual

kevancress, 2020-06-06:
Hi @HarryC1 thanks for the bug report.

A couple questions just to clarify things;

1) what in particular isn't working? are the dimensions not displaying, is their behavior not what you expect? 

2) Are there any errors in Blender's system console (can be found in the 'window -> toggle system console' menu) 

3) What version of MeasureIt-ARCH are you using?

4) What operating system and graphics card are you using?

If you can give me some more details I'll try and get the issue resolved!

kevancress, 2020-06-09:
Closing this for now, @HarryC1 Please reopen a new issue with the requested information above if you are still experiencing this issue

closed 2020-06-09 by kevancress