kevancress/MeasureIt_ARCH #92: Addon folder name

s-leger, 2020-06-28:
In order to be able to import into other packages in a reliable way, the root folder name of the add-on must never change, eg something like measure_it_arch will work, but importing MeasureIt-ARCH / MeasureIt.ARCH is impossible.
This also ensure that new version override older versions on update. 
We must rely on other means (like import the bl_info) to ensure a minimal version compatibility across modules.

To summarize, folder name must not use "-" "." or any special chars.
zip name may be whatever you want, including ". _ -" so eg MeasureIt-ARCH_0.0.4.zip is valid

kevancress, 2020-07-01:
Hey @s-leger thats good to know thanks!

Would using 'MeasureIt_ARCH' as the root folder name going forward work? I'll update the version 4 release zip with this name and keep it consistent for new releases going forward.

closed 2020-07-24 by kevancress

s-leger, 2020-07-25:
yes it will work like that, as far as i'm aware reserved chars are "." and non ascii