kevancress/MeasureIt_ARCH #93: Bug "MeasureIt-Arch Image" button is Freezing Blender

JLK67, 2020-06-30:
Dear, all
I hope I'm in the right place to raise this question, if not please accept my appologies.
I recently install MeasureIt-Arch 0.4
After trying to Render my measurments into an image the Blender application screens stay frozen.
Can someone advice me if this issue is already knowed, if yes how to solve it ?

Details,
The render picture is create, I see it under c:\temp\ folder, the message at the screen button says all was done, issues I loose the hand on Render, and I need to stop Blender using Task Manager End Task.
I used Blender V2.83.1 , MeasureIt-Arch 0.4 , MeasureIt-ARCH_v03_AppTemplate
PC is a Surface PC Processor Intel(R) Core(TM) i7-8650U CPU @ 1,90GHz  2.11GHz Ram 8.00 Go
Windows 10
Blender seems to works fine and fluidly with this configuration as long I stay in normal sice rendering this until I use the "MeasureIt-Arch Image" button
Thanks for your appreciated support any advice are welcome.
Cordially JLK

kevancress, 2020-07-01:
Hi @JLK67 

Thanks for the error report! I've experience this once or twice myself, but haven't been able to track down the cause yet. Sometimes I've found that left clicking with the mouse once in the 3D viewport after the render has completed restores responsiveness to blender, but clicking anywhere else causes it to freeze.

I'll look into what might be causing the issue and see if I can get it fixed!

sasinimal, 2020-07-04:
Dear  Sir,
I encountering the same problem. I am using Blender 2.83.1 and Measureit-Arch 0.4.1. 
Left clicking with the mouse once in the 3D viewport after the render has completed also does not restore responsiveness to blender.

Kindly look into this
Sasi

sasinimal, 2020-07-13:
Dear Sir,
Apart from the previous comment I sent on freezing of blender after rendering Measureit image, I list below some changes which would enhance the convenience and presentation of the measurements.

1. The annotation text field 1 and 2 are displayed too close in the view port that the bottom of the text gets clipped. Eg. g,q.

2. If I use fonts from the windows font folder the dimensions are not displayed. 

3. The axis selection for dimensions can be included in the style properties so that for each dimension the axis need not be changed from the default x-axis and can be done creating different styles for each axis.

4. While creating  dimension / annotation by default the style link can be activated with the first or the latest style created so that for each dimension / annotation the link button need not be enabled and the style selected.  

Hope these can be considered for your future updates.

globalcitizen, 2020-07-15:
I believe this is the same as #96.

JLK67, 2020-07-15:
Dear, Kevancress

First thanks for your fast reply.

Over the last 15 days I continue to use MeasureIt Arch, to create some blueprint.
I can confirm the issue appear from time to time.
I was trying to find some specific situation where application stops and freezing effect appear.
Unfortunately, I don’t find any repeatable case.

Your tool is great.
I reach my first target, even if it was with certain difficulties.
I wasn’t sure the issue, was related to a misusage or to my PC device.
I “feel” from time to time after having push rendering MeasureIt-Arch button, the PC performance slow down not obligatory crash.
Maybe some memories free-up, conflict….

I will continue and go further in my investigation, if I find something I will let you know.

May be this up coming suggestion are already mentioned somewhere in your different communication items, 
Even, I use this opportunity to highlight my user improvement suggestion,
Some functionalities could be maybe be developed, like
1. The possibility to move the dimension text around the measured line, not only left, right or middle but free positioning define by user.
2. Rotating the text will be helpful as well.
These suggestion are not mentioned because I like to have them "Nice to Have", but sometimes the MeasureIt-Arch tool create some display constraints in the drawing/blueprint creation intent, these due to placing measured dimension in a way or in an other automatically forcing to have space when measured dimension are close to each other, when rotating or moving the text will be enough.
These constraints force me to adjust/arrange the drawing/blueprint, sometimes with difficulties, your addon can be much more user friendly is you can fulfil these improvement challenges.

We stay in touch.
Thanks again for your great job.
Cordially JLK67

sasinimal, 2020-07-15:
Dear, Kevancress,
#96 is regarding show button of Measureit Arch. But this #93 is freezing of blender when Measureit Arch Image button is pressed ( under Measureit Arch Render of output properties Tab). The image gets generated and then blender freezes. (except at few occasions, it works OK)

kevancress, 2020-07-18:
Hi @JLK67 @sasinimal @globalcitizen 

Putting this issue as first priority for my work this weekend. If anyone is able to post a .blend file where this crash happens frequently that would be incredibly helpful in tracking down the issue. Also any info about your particular system, graphics card etc could be helpful.

Thanks
Kevan.

kevancress, 2020-07-18:
Also if anyone is able to run blender from the command line using 'blender.exe --debug-gpu' and could copy and paste the error result that occurs that would be much appreciated. Even without a crash I'm getting the following error after rendering a MeasureIt-ARCH Image, curious to see if others experience the same.

```
GL API error: GL_INVALID_OPERATION error generated. The required buffer is missing.
Stack trace:
blender.exe         :0x00007FF7B06D2210  bli_windows_system_backtrace_stack_thread
blender.exe         :0x00007FF7B06D1480  BLI_system_backtrace
blender.exe         :0x00007FF7B064C070  gpu_debug_proc
nvoglv64.dll        :0x00007FFCAB6AE3A0  Symbols not available

 GL API performance: Pixel-path performance warning: Pixel transfer is synchronized with 3D rendering.
Info: New image created with measures. Open it in UV/image editor
```

or if you run the 'blender_debug_gpu.cmd' file that is present along with your blender install it will export the error log to a text file that I can sift through if you're able to post it

sasinimal, 2020-07-18:
Dear, Kevancress,
Based on your request, I ran blender in the cmd window with the sugested command line. The Measureit image got generated. then when I pressed render menu for the scene, there was no response. on repeated pressing blender crashed and came to cmd window. I copied the dump on the cmd window which is attached below as Error 1. 
[Error 1.txt](https://github.com/kevancress/MeasureIt-ARCH/files/4941054/Error.1.txt). The corresponding screen shot after Measureit image got generated is: 
[image]

Then I repeated opening blender the same way and Generated Measureit image. When I pressed Render menu for scene, there was no response. Then When I pressed Scripting menu it opened and I could see some messages. Also the Python console heder was flickering (as marked. This type of flickering is seen in warious occations of generating the Measureit image.)
[image]

After closing Blender from windows Task manger, The CMD window showed the following. (I think no error message is shown).

I am also attaching a drop box link to the Blender file where I encountered the problem. Hope you can view the file.
https://www.dropbox.com/scl/fi/of9c2k6to4c6rixqqgjz1/Nandanam-Home-Dimnsions-GRound.blend?dl=0&oref=e&r=ABNkuaZVyH7jC1YIH5sGauzcBCvUIZ5OMGgNQ32DkDotxzSLg-ujwk73fG78JEPjUGrCPlcCXJ5y155eS-acEKeyjb6XPqLcKFZjeZ1k-m7VngzCHuoY4PhOOgd4dt11tokNrADLJgqQEvGgaoaiG0e3VcGazyQ9fJ-VMxCEtcMA3D-wqLu632hY4-pex5UpWtmyt81BMVnLjKsWoxpolX1A&sm=1

Hope this might help you in your resolving the issue.
Sasi.

kevancress, 2020-07-18:
@sasinimal This is fantastic thank you! I've been able to reproduce the issues you're having with this file and there's some interesting error codes in that text file that should give me a good place to start looking. Thanks!

JLK67, 2020-07-18:
Hi,
Thanks to all for your efforts you are deploying to fix the issue.
I can now reproduce the issue as well.
I create a very basic cube with 3 dimensions measured.
After rendering by using MeasureIt Arch button my screen are freezed.
Attached you will find the "blender_debug_output" & "blender_system_info" file generate by starting 'blender_debug_gpu.cmd'
You will find as well the Blender file in a Zip format.
Unfortunatly I can't execute the first of your option 'blender.exe --debug-gpu' (I don't know how as I'm not expert)

Cordially JLK67
[image]

[blender_debug_output.txt](https://github.com/kevancress/MeasureIt-ARCH/files/4941549/blender_debug_output.txt)
[blender_system_info.txt](https://github.com/kevancress/MeasureIt-ARCH/files/4941559/blender_system_info.txt)
[Sample MeasureIt Arch 01.zip](https://github.com/kevancress/MeasureIt-ARCH/files/4941575/Sample.MeasureIt.Arch.01.zip)

sasinimal, 2020-07-19:
Dear Kevancress,
Thank you for starting to  resolve the the error. Pl. let me know if you need any more help.

kevancress, 2020-07-19:
@sasinimal one more piece of info that could be helpful. What graphics card are you running?

It wouldn't happen to be a NVIDIA GTX 1050 would it? I've just noticed that both @JLK67 and I are using the same card, and although I don't think its likely to be the cause of this issue it is curious...

sasinimal, 2020-07-19:
Its Nvidia GeForce GTX 1060.

globalcitizen, 2020-07-20:
FWIW unclear at this point if it's the same bug or not, but I am on "Intel UHD Graphics 630" on Mac.

Toss90, 2020-07-20:
I am on "Quadro 2000" on win 10 1903

kevancress, 2020-07-20:
Hey All,

Thanks for the help on this. I still haven't managed to work out the source of the issue, but I've reached out to the blender dev community over on devtalk to see if I can get some help on this. (Thread here: https://devtalk.blender.org/t/drawing-to-gpuoffscreen-from-within-an-operator-seems-to-freeze-blender-until-the-3d-view-is-redrawn/14459)

As far as I can tell, the offscreen buffer I'm using to render MeasureIt-ARCH elements seems to be conflicting with the UI's texture allocations or something. Every time I've reproduced the issue, forcing the 3D view to redraw has restored functionality to Blender. I haven't found a way to do this automatically through the add-on yet, but holding middle mouse and panning around in the 3D viewport seems to restore normal functioning 9 times out of 10. 

[image]
 

Gonna keep digging and see if I can get a better fix, hopefully something comes out of the dev talk thread, I'll keep ya posted.

sasinimal, 2020-07-21:
Dear Kevancress,
Thanks for your efforts. How ever my feed back for redrawing the view port  is that it did not work for me. After generating image , a message was flashings but no mouse action including MMB was working on the view port.
[image]

sasinimal, 2020-07-21:
One more info, when trying to close Blender using Task manger, 2 programs were shown in the list, but mouse over on task bar was showing only the main blender window. This may be helpful for you.
[image]

kevancress, 2020-07-24:
Alright, so after much attempted debugging and confusion it turns out that the root of this issue was much simpler than I expected. 

Certain render cases weren't disabling openGL depth testing, which then caused the UI to freeze up and not draw properly. I've committed a fix in the main master branch 93a03ab0823618a66ae84b6e96b06ac9b38d9aa7 and also updated the version 0.4.5 release and it seems to have resolved all of the freezing for me, even on the test files you all provided!

If you could download the latest version, test it out and let me know if the fix works for you that would be fantastic!
Thanks!

JLK67, 2020-07-24:
Hello
Thanks it seems you get it, but I can't tell you.
After downloading the zip file, I try to activate version 0.4.5
Unfortunatly I get following message after trying to select the field "enable the addon", in menu Edit, Preference.
Did I make something wrong ?

[image]

kevancress, 2020-07-24:
Ah sorry about that @JLK67  thats an issue with an external library I'm using for svg export. I've just fixed it so if you redownload the release file and re-install the new version it should be working now

sasinimal, 2020-07-25:
Dear Kevancress,

Thank you for resolving the freezing issue. Now Measureit Arch and Blender are working fine. I also thank you for considering my feed backs on this issue.

For further improvements,  I list below some changes which would enhance the convenience and presentation of the measurements. The first point is really important.

1. The annotation text field 1 and 2 are displayed too close in the view port that the bottom of the text field 1 gets clipped. Eg. g,q.
2. If I use fonts from the windows font folder the dimensions are not displayed.

3. The axis selection for dimensions can be included in the style properties so that for each dimension the axis need not be changed from the default x-axis and can be done creating different styles for each axis.

4. While creating dimension / annotation by default the style link can be activated with the first or the latest style created so that for each dimension / annotation the link button need not be enabled and the style selected.

Hope these can be considered for your future updates.
Thanks again,
Sasi.

JLK67, 2020-07-25:
Hello,

Thanks Kevancress, for your engagement, and for solving the issue.
It works fine for me as well.

From my point of view we can close this point, and concidered as solved.

I will continue my project that will take me a while, in the meantime, I hope to get some improvement as mentionned in previous discussion.

**GREAT JOB !**

Cordially JLK67

closed 2020-07-25 by JLK67