kevancress/MeasureIt_ARCH #98: 0.4: Overlapping annotation handling (offsets)

globalcitizen, 2020-07-11:
Here is an example where two overlapping objects have had bounds dimensions added along the same axis and orientation.

Their centers differ so both dimensions remain visible, however they run together leading to possible misunderstandings.

Ideally they should automatically be placed at differing offsets from the edges they are measuring to preserve clarity.

[image]

kevancress, 2020-07-16:
Hey @globalcitizen 

Sorry about the slow replies recently. Been a bit busy with work. 

Hmm thats a good point. I might be able to solve this by checking the other dimensions on creation and adjusting the offset accordingly, like you say. I'll see if I can get it worked out this weekend!

globalcitizen, 2020-07-17:
No worries and congrats on the new release! Awesome work!