kevancress/MeasureIt_ARCH #99: v0.4 Annotations lower-case descenders cropped (macOS)

notbyhalf, 2020-07-12:
Hi,

Thanks for this plug-in and I apologies if this "issue" is just me not using it properly. It seems that lower-case characters with descenders in annotations are cropped at the underline.

For example, I have the annotation line thickness set to 0 and am labelling rooms on a floor plan. "Study" appears as "Studv" because the descender on the "y" isn't showing, and "Dining Room" also has the bottom of the "g" cut off. 

It seems to happen with both Text Field 1 and Text Field 2 and I couldn't find a setting to adjust vertical placement relative to whatever is cropping the text. Position and Alignment move the text but it doesn't change the cropping.

cheers,
Laurence

kevancress, 2020-07-16:
Hi @notbyhalf 

Definitely an issue, thanks for reporting! I've been playing around with the text placement settings and thought I'd gotten this issue fixed (it is on windows) but I may have inadvertantly broken it on mac. I'll take a look this weekend and see if I can get it worked out

kevancress, 2020-11-07:
@notbyhalf Sorry I never replied directly too this, I've made several changes to the text drawing system in the last few months so this should be fixed. Please re-open the issue and let me know if your still having this issue when using the master version!

closed 2020-11-07 by kevancress