kevancress/MeasureIt_ARCH #100: SVG output contains only Line Groups and Aligned Dimensions using 0.4.1 (dev)

meikasahara, 2020-07-13:
Blender version: 2.90.0 Alpha, branch: master, commit date: 2020-07-10 20:07, hash: d2b910fafec6, type: Release
build date: 2020-07-10, 23:06:55
platform: Linux
----------------------------------------------------------------------------------------------
 Using version 0.41 (Dev) of MeasureIt ARCH, downloaded (cloned) directly from github, only Line Groups and Aligned Dimensions appera in the svg output file after rendering. Using the regular MeasureIt Image render button, the png (set by default in blender) render output shows every measurement and annotation.
Below are the png render output, a screenshot of the Scene setup and the svg output, respectively.
[image]
[image]
[image]

kevancress, 2020-07-16:
Hey @meikasahara  Thanks for the report!

This is a known issue for now, The SVG export is still very much experimental and in development, but I'm definitely working on getting the rest of the Dimension Annotation and Line types supported!

meikasahara, 2020-07-17:
One additional notice is that it would make the appearance of multiple measurements cleaner if the feature of automatically shifting the text left or right if it is larger than the space between the dimension lines could somehow be controlled (on or off). 
Thanks for the reply :)

kevancress, 2020-07-24:
Closing this for now, but I've opened up a new issue for the auto placement toggle 

Also I've hidden the SVG export behind an 'enable experimental' toggle in the MeasureIt-ARCH scene settings for now until it's a little bit more feature complete.

closed 2020-07-24 by kevancress