kevancress/MeasureIt_ARCH #105: Missing svgwrite

s-leger, 2020-07-23:
Hi, 
Looks like the svgwrite module does not ship with python by default. 
Blenderbim ship with this module - but you should not rely on other addon deps - and looks like MeasureIt-ARCH is not able to properly import from blenderbim.

```
File "/home/xxx/.config/blender/2.83/scripts/addons/MeasureIt-ARCH/svg_shaders.py", line 31, in <module>
    import svgwrite
ModuleNotFoundError: No module named 'svgwrite'
```

Moult, 2020-07-23:
https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.6.0/src/ifcblenderexport/Makefile#L85 FYI

kevancress, 2020-07-24:
Hey @s-leger  & @Moult 
Thanks for catching this, I'll include the svgwrite module and update the v0.4.5 release today.

@Moult do you know if there is a quick way to synchronize the inclusion of external libraries directly through github rather than by running a make file?

Also referancing @r3sonant's issue here since it's the same problem
https://github.com/kevancress/MeasureIt-ARCH/issues/104#issue-664229395

kevancress, 2020-07-24:
Alright, this should be fixed in fa91f946110425f707c6b4804390b160b3cb6a0f and I've updated the .zip for the version 0.4.5 release as well

kevancress, 2020-07-24:
Re-Opening because I forgot to include pyparse.py

kevancress, 2020-07-24:
pyparsing.py added in 24455dda846a146a33605d227d8299c9cdae4978

closed 2020-07-24 by kevancress

s-leger, 2020-07-25:
Hi Kevan,
the nested /site/package/ structure to store modules in blenderbim was required by opencascade static relative path in build expecting the .dll / .so to be in a ../../bin folder.
It is not required to support svgwrite, so you may simply replace nested /site/package by something like /lib

kevancress, 2020-07-26:
Ahh good to know! Thanks @s-leger I'm still pretty new to including external libraries (and really python in general) so tips like that really help!