kevancress/MeasureIt_ARCH #107: 0.4.5: Dimension auto-flip imperfect

globalcitizen, 2020-07-25:
First of all, great job overall on the recent updates! In general the dimensions are far more legible now.

However, moving close to an axis appears to flip the dimensions.

Higher up position...

[image]

Lower down position...

[image]

As this flip does not appear to be useful, I wonder if there is a means to rectify it.

kevancress, 2020-07-26:
Ahh Looks like I need to fine tune my text flipping thresholds! I'll try and take a look at this today, doing quite a bit of refactoring there  to set up for more direct control of text placement and to make it easier to add in scale based text sizing soon so I'll try and make sure this gets fixed too

kevancress, 2020-07-26:
Should be fixed in 9ebcdeb701e43867beb72c53a0aa4daf7d7483b6

closed 2020-07-26 by kevancress

globalcitizen, 2020-07-28:
Sorry for the dumb question but how do I get that latest version installed to test with?

I tried checking out your repo with git then zipping similarly structured to the released zipfile (ie. in a MeasureIt_ARCH subdir) but going through the install button yields an error as follows.

[image]

I am happy to do a symlink or something from the installed version but wonder what is best practice here. I cannot find any clear Blender-world reference on how to keep your blender module that is checked out from git at the bleeding edge.

Might be worth considering adding this to the docs, also, since releases seem sporadic.

globalcitizen, 2020-07-31:
Should I add the above as a separate issue?

globalcitizen, 2020-08-07:
Added to [general suggestions issue](https://github.com/kevancress/MeasureIt-ARCH/issues/4).