kevancress/MeasureIt_ARCH #109: Huge font sizes and arrows too small compared to text

Moult, 2020-08-03:
See attached file:

[measureit.zip](https://github.com/kevancress/MeasureIt-ARCH/files/5015496/measureit.zip)

I'm not sure what I did wrong, but I can't seem to control font sizes and arrows particularly well. I've set a dimension style, which seems to be consistently applied for single objects, but when I annotate between two points on two separate objects (which needs to happen), I get very large text. You can also see below that the arrows are nearly invisible.

[image]

Sorry for the poor explanation, hope the images help.

kevancress, 2020-08-10:
Hey @Moult, I think this should be resolved in the latest master.

The issue was because my old text sizing system had no meaningful output to reference when creating dimensions, so the font size was proportional to the distance being measured (which was a bit hack-y and not super useful)

Now that I have views set up and font size is defined in meaningful units (in pt's where 1 pt = 1/72") the default font size is 18 pt's now or 1/4" which is fairly standard, in Canada at least. This size is maintained based on the views scale, or if no view is specified a default scale of 1:25 is used (picked because it works well with blenders default scene, but it can be adjusted in the unit settings) 

I've also been tweaking the default arrow sizes, they should also be defined relative to the output scale now too. Let me know if the latest is working alright, or if there are any more changes I can make to make things more standard / professional

Moult, 2020-08-12:
Confirmed fixed. Awesome and thanks!

closed 2020-08-12 by Moult