kevancress/MeasureIt_ARCH #110: Improvements in BlenderBIM integration

Moult, 2020-08-03:
So we have a thing :)

[image]

Would it be possible for https://github.com/kevancress/MeasureIt-ARCH/blob/master/measureit_arch_external_utils.py#L30 to return the actual offset locations, otherwise they appear exactly on vertices as shown?

Also returning the coord list might be nice :) https://github.com/kevancress/MeasureIt-ARCH/blob/master/measureit_arch_external_utils.py#L81

kevancress, 2020-08-10:
Hey @Moult, Sorry about the slow reply, I've been off in the woods without an internet connection for the last week!

commit f58ba270dac08b8b37cd794db5e5bc849e37257d now returns the coord list (sorry that was silly of me to just print it ) and adds an optional argument offset_pos (true by default) to specify that the function should return the start and end points of the dimension line rather than the anchor points.

I'll try and clean this up and generalize it to all of the dimension types this week too!

Moult, 2020-08-12:
Hmm, I added a dim to a wall with a solidify modifier (5.90m in the screenshot) and it returned an empty list.

Then I added a cube with a 2m dimension, and I got an exception:

[image]

kevancress, 2020-08-12:
Ahh, right, sorry some of my drawing code makes the assumption that it's always occurring in the context of a 3D viewport, which I suppose isn't the case when it's being called externally.

Should be fixed in 88fa4920ff9911e0dbc93dcba50240b532c07efb, there does need to be a 3D viewport open somewhere in the blender workspace, but as long as one is present there should be no issue

[image]

Moult, 2020-08-13:
Hmm, the dimensions still don't seem to be offset: 
[image]

I call it in my script like so:

```
    def draw_measureit_arch_dimension_annotations(self):
        try:
            import MeasureIt_ARCH.measureit_arch_external_utils
            coords = MeasureIt_ARCH.measureit_arch_external_utils.blenderBIM_get_coords(bpy.context)
        except:
            return
        for coord in coords:
            self.draw_dimension_annotation(Vector(coord[0]), Vector(coord[1]))
```

When calling it in my script in the operator, I get these coords printed:

```
[[Vector((-1.0, 1.0, 1.0)), Vector((1.0, 1.0, 1.0))], [Vector((1.0, -1.0, 1.0)), Vector((1.0, 1.0, 1.0))]
```

As you can see, they are coincident with the verts of the default cube.

However, if I call it within the Python console, then indeed I get the offset coordinates:

```
>>> MeasureIt_ARCH.measureit_arch_external_utils.blenderBIM_get_coords(C)
[[Vector((-1.0, 1.5, 1.0)), Vector((1.0, 1.5, 1.0))], [Vector((1.5, -1.0, 1.0)), Vector((1.5, 1.0, 1.0))]]
[[Vector((-1.0, 1.5, 1.0)), Vector((1.0, 1.5, 1.0))], [Vector((1.5, -1.0, 1.0)), Vector((1.5, 1.0, 1.0))]]
```

I don't understand why the discrepancy?

An additional issue, perhaps unrelated, is that I get coordinates for aligned dims, but not axis dims.

kevancress, 2020-08-15:
hmm, there's an optional argument 'offset_pos' that should be true by default, but it might fix things if you set it to true explicitly?
` coords = MeasureIt_ARCH.measureit_arch_external_utils.blenderBIM_get_coords(bpy.context, offset_pos = True)` might do it. I'll look into it further though.

And yep right now I've only got this set up for aligned Dims. I'll try and get it set up for the other types soon.