kevancress/MeasureIt_ARCH #115: Git Version: Errors when rendering ARCH-Image and Vector

meikasahara, 2020-08-13:
1) When rendering an image using the ARCH-Image render button, the following error appears:
Error: Python: Traceback (most recent call last):
  File "/.config/blender/2.91/scripts/addons/MeasureIt-ARCH-master/measureit_arch_render.py", line 129, in execute
    render_result = render_main(self, context)
  File "/.config/blender/2.91/scripts/addons/MeasureIt-ARCH-master/measureit_arch_render.py", line 258, in render_main
    renderoffscreen = gpu.types.GPUOffScreen(width, height, samples = 0) #
TypeError: GPUOffScreen.__new__() takes at most 2 arguments (3 given)

If line 258 of measureit_arch_render.py is changed from
    renderoffscreen = gpu.types.GPUOffScreen(width, height, samples = 0)
to
    renderoffscreen = gpu.types.GPUOffScreen(width, height)
this error disappears and the image can be normally rendered. 

2) When rendering an svg using the ARCH-Vector render button, the following error appears:
Error: Python: Traceback (most recent call last):
  File "/.config/blender/2.91/scripts/addons/MeasureIt-ARCH-master/measureit_arch_render.py", line 230, in execute
    if render_main_svg(self, context) is True:
  File "/.config/blender/2.91/scripts/addons/MeasureIt-ARCH-master/measureit_arch_render.py", line 540, in render_main_svg
    if view.res_type == 'res_type_paper':
AttributeError: 'NoneType' object has no attribute 'res_type'

If lines 540-542 are commented, this error disappears and the render proceeds normally.

3) Also, there is an inconsistency when adding annotations in Edit mode, as these cannot be added without having previously added an annotation in object mode which creates an empty at the 3d cursor location. After this object mode annotation has been created, the edit-mode annotation can be created normally.

kevancress, 2020-08-15:
Hey @meikasahara thanks for the report! I hadn't tested on 2.91 yet so thanks for the heads up! I'll fix these this afternoon

closed 2020-08-15 by meikasahara

kevancress, 2020-08-15:
Should be fixed in 7d2966c39b60747a6803b336d8df98cac83403bd

meikasahara, 2020-08-15:
Yes, I tested it and it works, so I closed the issue :) Thanks