kevancress/MeasureIt_ARCH #116: Find a way to clean up shader code

kevancress, 2020-08-16:
There's a lot of duplication of code in the current shader set up that could use some cleanup.

(This might include depreciating support for < 2.82) unless I can find an easy way to swap out the use of the blender_srgb_to_framebuffer_space() function depending on the blender version... because 2.82 won't compile the shader if that function is present.

meikasahara, 2020-08-17:
I don't know if this is relevant to this clean up of the latest commit in shaders.py file, but with this in place the, Dot end cap in Annotations is no longer visible in the viewport or the render output. The arrow still works though. (tested with latest blender 2.91 alpha on linux). Reverting the shaders.py to an earlier version resolves this.

kevancress, 2020-08-17:
@meikasahara  Yep thanks for the heads up, looks like I broke my point shader while I was messing around with the shader code. Should be fixed in 8e4cc3dae3fd84e51439963069ed1b4051797639

kevancress, 2022-12-27:
Finally getting back to cleaning this up. Currently in the draw_rewrite branch, All shaders are now loaded from .glsl files, which should make editing and debugging simpler

4b016aafe7647d62af9c48b73492d4a9d77d310f

Working on a single general purpose shader for all line types