kevancress/MeasureIt_ARCH #120: A way to Dimension Roof Slopes

kevancress, 2020-08-23:
Need a way to show slope info. Could be an option in the angle dimensions (degrees/radians/slope?) or it's own dimension type

globalcitizen, 2020-08-23:
In Australia it is regulated in degrees.

kevancress, 2020-08-23:
@globalcitizen 

That's interesting to know that its specified in degrees in Australia, It might make sense for me to keep this associated with the angle dimensions then, as they already display degrees.

Here in Canada the fractional notation is more common (5:12 is a common roof sloop, 1:20 is common for ramps etc.)

globalcitizen, 2020-08-23:
Out today but will send some example legislation later.

Moult, 2020-09-08:
In Australia, roofs use degrees, but ramps use ratios :) A funny world, isn't it!