kevancress/MeasureIt_ARCH #121: Recursion Error  and Blender freeze/crash with Custom Shape Annotations in Collections

meikasahara, 2020-08-25:
Platform: Linux
Using Blender 2.90 Beta (83c78529b9ef), using the latest git version of MeasureIt_Arch, when attempting to add a custom shape to an annotation, on selecting the collection with the object for custom shape, Blender becomes unresponsive and the terminal goes haywire with rapid succession of messages such as the following, in which "RecursionError: maximum recursion depth exceeded while calling a Python object" frequently appears. 

```
  File "~/.config/blender/2.90/scripts/addons/MeasureIt_ARCH-master/measureit_arch_geometry.py", line 3427, in draw3d_loop
    draw_annotation(context,myobj,annotationGen,mat,svg=svg)
  File "~/.config/blender/2.90/scripts/addons/MeasureIt_ARCH-master/measureit_arch_geometry.py", line 2287, in draw_annotation
    draw3d_loop(context,objs,svg=svg,extMat=mat)
  File "~/.config/blender/2.90/scripts/addons/MeasureIt_ARCH-master/measureit_arch_geometry.py", line 3427, in draw3d_loop
    draw_annotation(context,myobj,annotationGen,mat,svg=svg)
  File "~/.config/blender/2.90/scripts/addons/MeasureIt_ARCH-master/measureit_arch_geometry.py", line 2287, in draw_annotation
    draw3d_loop(context,objs,svg=svg,extMat=mat)
  File "~/.config/blender/2.90/scripts/addons/MeasureIt_ARCH-master/measureit_arch_geometry.py", line 3427, in draw3d_loop
    draw_annotation(context,myobj,annotationGen,mat,svg=svg)
  File "~/.config/blender/2.90/scripts/addons/MeasureIt_ARCH-master/measureit_arch_geometry.py", line 2280, in draw_annotation
    draw_lines(lineWeight,rgb,coords, twoPass=True,pointPass=True)
  File "~/.config/blender/2.90/scripts/addons/MeasureIt_ARCH-master/measureit_arch_geometry.py", line 3256, in draw_lines
    draw_points(lineWeight,rgb,pointCoords,offset)
  File "~/.config/blender/2.90/scripts/addons/MeasureIt_ARCH-master/measureit_arch_geometry.py", line 3187, in draw_points
    batch = batch_for_shader(pointShader, 'POINTS', {"pos": coords})
  File "~/LOCALSSD/blender-2.90-83c78529b9ef-linux64/2.90/scripts/modules/gpu_extras/batch.py", line 37, in batch_for_shader
    from gpu.types import (
  File "<frozen importlib._bootstrap>", line 1019, in _handle_fromlist
RecursionError: maximum recursion depth exceeded while calling a Python object
```

kevancress, 2020-08-25:
Ahhhhh Yep that makes sense, I need to prevent an annotation being in the collection that defines its custom shape, otherwise it will just keep trying to draw itself recursively... 

I've only been defining my custom shapes in a separate scene, and hadn't thought how easy it is to accidentally include an annotation in its own custom shape collection. I'll fix this tonight

kevancress, 2020-08-25:
Alright, the worst of this is fixed in e148dc3f1e5bf22d7714928b5adedf0230a4ed52, but I still want to add a poll method to the collection search box, so that the collection that the annotation is a part of wont even be selectable

closed 2020-08-26 by meikasahara