kevancress/MeasureIt_ARCH #125: Error when adding dynamic line groups or line groups by crease

meikasahara, 2020-09-08:
Blender Version: 2.91.0 Alpha, branch: master, commit date: 2020-09-07 22:02, hash: 5c2ac8520e07, type: Release
build date: 2020-09-07, 23:03:13
platform: Linux

1) When adding dynamic line groups or line groups by crease to an object such as Icosphere or UV Sphere, no lines appear on the viewport (although in some occasions it works for a cube), and the console repeatedly (at least 4 times) displays the following error:
`Traceback (most recent call last):
  File "~/.config/blender/2.91/scripts/addons/MeasureIt_ARCH-master/measureit_arch_main.py", line 583, in draw_callback_3d
    draw_main_3d(context)
  File "~/.config/blender/2.91/scripts/addons/MeasureIt_ARCH-master/measureit_arch_main.py", line 543, in draw_main_3d
    draw3d_loop(context,objlist)
  File "~/.config/blender/2.91/scripts/addons/MeasureIt_ARCH-master/measureit_arch_geometry.py", line 3432, in draw3d_loop
    draw_line_group(context,myobj,lineGen,mat,svg=svg)
  File "~/.config/blender/2.91/scripts/addons/MeasureIt_ARCH-master/measureit_arch_geometry.py", line 2124, in draw_line_group
    lineBatch3D[batchKey] = batch_for_shader(lineGroupShader, 'LINES', {"pos": coords,"weight":tempWeights})
  File "~/LOCALSSD/blender-2.91.0-a/2.91/scripts/modules/gpu_extras/batch.py", line 55, in batch_for_shader
    vbo.attr_fill(id, data)
ValueError: Expected a component of size 3, got 1`

2) Sometimes the above error does not appear (for instance, when adding an annotation first and then a line group by crease to an icosphere). However, the following inconsistency has been noticed: when using an icosphere (without the faces) as a custom annotation object and then subsequently adding a line group by crease to the icosphere, a mirror image of the icosphere with the line group appears misplaced, as can be seen in the following screenshot of the attached blend file.
[image]

[measureit_error_blend.zip](https://github.com/kevancress/MeasureIt_ARCH/files/5191349/measureit_error_blend.zip)

kevancress, 2020-09-14:
Hmmm, okay I think I've got a sense of what's causing this. For custom shapes I'm drawing the geometry of the custom shape object directly, mapping the object origin onto the anchor point (actually in the latest version it's on the text anchor but... well there's a whole other set of errors with rotation that that's causing... ), and I'm also drawing any line groups attached to that geometry, but mapping the world origin onto the anchor point... hence the apparent offset/duplication... 

I'll do some testing today and see if I can work up a system that's a bit more consistent / intuitive

The first error likely has somthing to do with how I'm caching line groups, really I need to overhaul that whole system to make the dynamic line groups work properly, so lots of changes to come there

kevancress, 2020-09-14:
Alright, these should be resolved in 400830d9fe51c8b3e5ae09dc7de65398fdeaca98 and 78a725fe05f7f966a43aea42723587f698d9ce1a

For the first issue, the errors were occurring because there were no lines to draw, the default crease angle for the dynamic line groups is 30 degrees which doesn't result in any lines on a uv sphere, so you'll need to lower the crease angle to see any results (for performance, the dynamic crease only updates when entering and leaving edit mode, or when the evaluate depsgraph check box is selected, so you'll need to do either of those too before you see the updated lines... I'm working on a more frequent update, but the dynamic line code is still way too slow right now)

meikasahara, 2020-09-14:
Excellent work! I can confirm that the latest commits resolve the above issues, thank you very much!

closed 2020-09-14 by meikasahara