kevancress/MeasureIt_ARCH #127: How can I delete added dimension?

Diomas, 2020-09-15:
Sorry for a dumb question didn't find it in docs

MeasureIt ARCH adds `Add Dimensions` panel so I can add ones, but I don't see where can I select/remove created dimensions?

meutzitzu, 2020-09-15:
you go into the properties window, and under Object you scroll down to find "MeasureIt-ARCH Dimensions"
you can scroll between them and toggle each one's visibility till you find the one you want, then hit the X button on the right 

This is very very stupid, of course. My deepest apologies if there's an easier way to do this

my suggestion would be to add a "show selected" feature ... basically, whenever you click on a dimension in the properties tab, and the contextual settings such as Plane, Distance, Offset etc appear below, it colors the "active" dimension with the Blender Selection color (kinda like how it is with MeasureIT-ARCH Lines) 

this will solve most of the frustration and would be reasonably easy to Implement, I assume
I would also suggest a "picker" function where you add a picker icon, upon which you click, suddently all "Gizmos" for the visible dimensions appear, but clicking one won't move the distance, it will just select that particular dimension in the list.

kevancress, 2020-09-15:
@Diomas No dumb questions, issue like this really help me work out where I need to put more work in the User Experience side of things for the add on!

Here's a quick gif of the process for deleting the dimensions.

[image]

To add to what @meutzitzu said above, MeasureIt-ARCH does have a gizmo for some dimensions, and using the gizmo will cause that dimension to become selected in the list as you can see above.

@meutzitzu Having the active dimension in the list be colored with Blenders selection color is a fantastic idea! can't believe I hadn't thought of that haha. I'll try and add that in this week, going to re-name this thread as a reminder if that's alright?

kevancress, 2020-09-19:
alright! I've added this in. Had to refactor some of the underlying UI code, which will break some backwards compatibility so it's in a new branch called 'version_0.5' f5fe1b9f061f80505d4ad2a77e3ef99b73b84470

[image]

closed 2020-09-19 by kevancress