kevancress/MeasureIt_ARCH #128: **Proposal: Vectors instead of floats for Offset and Rotation in Dimensions settings.**

meikasahara, 2020-09-16:
**Proposal: Vectors instead of floats for Offset and Rotation in Dimensions settings.**
Using 3d vectors and euler rotations in Dimension settings could perhaps enable the user to offset and rotate the measurement more freely, ( maybe when Auto-Placement is disabled) similarly to annotations.
Edit: Something similar would be to enable the rotation of a measurement with respect to the normal vector of the view plane or of the plane on which the dimensions lies, as well as the option to slant a measurement with respect to the normal vector of the measured edge (perhaps using a custom object, as in Axis Dimensions). The latter functionality is present in the original MeasureIt addon:
[image]

_Originally posted by @meikasahara in https://github.com/kevancress/MeasureIt_ARCH/issues/4#issuecomment-680951546_

meutzitzu, 2020-09-16:
I think one of the main design decisions of MeasureIt-ARCH was that you should use styles that you define once in the file, and apply them to individual dimensions instead of having to tweak the position of every single dimension. 
I don't see how the viewplane & distance & rotation does not cover all use cases

meikasahara, 2020-09-16:
Of course the default behavior should be as it is now but I was mainly thinking about having the extra option to allow the text to be rotated with respect to the view plane in addition to the dimension axis, which might be useful in some cases (avoiding cluttering and line overlaps).